cocotb/cocotb #1080: Testbench checking with a mock simulator?

cmarqu, 2019-08-21:
One of the main differences between cocotb and traditional verification setups is that the cocotb testbench may have a runtime error that only shows up after simulating to this point, which may be hours or days later. I remember (but cannot find again) a recent discussion about cocotb vs. SV and this was the major disadvantage for some people.

I wonder if we could find some of these problems early. One thing that came to mind is having some kind of a mock simulator that "immediately" returns from a ``Timer`` yield, immediately triggers a trigger, ...
This would not cover all code paths, but may still be useful. Or is there a way to statically check the code for (some of) the potential problems?

eric-wieser, 2019-08-21:
Adding `mypy` annotations could help here

themperek, 2019-08-22:
A good editor/linter (for python plenty exist) can avoid silly mistakes and for more complex, I would not be so sure SV is better.

eric-wieser, 2019-08-22:
Note that to enable users to get the most out of the `mypy` linter, cocotb would have to opt in and provide the necessary annotations.  Probably something that would be much easier after a python 3 transition.

ktbarrett, 2019-10-16:
In all but the simplest testbenches, triggers firing immediately would not cover all code: maybe A needs to trigger before B for a particular code path to be followed. And what results would you get back for reading a signal? Random is the only reasonable answer and it is not a good idea.

A part of the solution to the issues that Python brings to the table is the responsibility of the user. They should perform adequate internal checks, keep things simple to keep the error surface low, and reuse known good code.

ktbarrett, 2019-10-16:
But keeping on the mocking idea... Ideally, you would only need to mock out the triggers and the dut object while being able to use the same user-defined tests, regression system, and coroutine system. Then you could replace the simulator with anything: something to help you work out errors in complex testbenches, software models, a client-server abstraction layer to run tests out of process, etc. But first we must define a simple interface for this purpose and refactor everything else in the repo to build off this interface.

cmarqu, 2020-05-11:
Superseded by other PRs.

closed 2020-05-11 by cmarqu