cocotb/cocotb #1081: Add support for integration with IPython's "autoawait" feature

wimiko, 2019-08-22:
At the moment, it is not possible to run cocotb coroutines from an IPython shell. 
IPython is able to run asynchronous code (see https://ipython.readthedocs.io/en/stable/interactive/autoawait.html), but support for this has to be added to the scheduler.

ktbarrett, 2019-10-18:
Should this a scheduler "feature", or could we come up with something cleaner?

We could modify cocotb's start up so that once loaded cocotb will jump to a user specified entry point (we already do this sort of by passing cocotb the test module), which could be the regression manager, or could be IPython, or whatever. Ideally, the scheduler should be a stand-alone component: I should be able to use the scheduler in a purely Python environment. At that point, what we are trying to achieve is trivial.

Everything in cocotb is just too tightly coupled for my tastes.

eric-wieser, 2019-10-18:
I think step 1 is decoupling the concept of tests (ie, the regression manager) from the scheduler. I made a start on simplifying the interface, but they're still couple's.

eric-wieser, 2020-04-09:
Tried to make a start on this, but I can't work out how to set the hook:

https://gitter.im/ipython/ipython?at=5e8f21e09316f34b8d8ebed3

eric-wieser, 2020-04-13:
Alright, I have this working!

PR #1649 (2020-05-05): Add support for launching IPython
Closes #1081

Also need to decide where this should live.
* Should it be an extension, `cocotb_ext.ipython`?
   * How about  living in the extension namespace, but distributed as part of cocotb itself?
   * If not an extension, how should we handle the dependency on `IPython`?
* Should I split this into `ipython_runner` and `ipython_support`, where the former is used in a `MODULE=cocotb.ipython_runner` type example, while the latter is used for providing general tools like the custom prompt options I added here?

You can test this today by pasting the contents [... cut]
closed 2020-05-05 by eric-wieser