cocotb/cocotb #1083: Modelsim-FLI: cocotb does not drive signals correctly using the FLI

higgns, 2019-08-23:
This is related to #1059 and may be its root cause. I'm not very firm with C++ and the way cocotb uses the interface, so I didn't manage to fix this.

To set a signal value over the FLI cocotb uses `mti_SetSignalValue()`. The FLI reference states:

> Setting a resolved signal is not the same as driving it. After a resolved signal is set it may be changed to a new value the next time its resolution function is executed. mti_ScheduleDriver() and mti_ScheduleDriver64() can be used to drive a value onto a signal.

If I understand it correctly, the correct way to drive signals from cocotb would be to create a driver using `mit_CreateDriver()` and store its handle. Once we want to change its driven value, use `mti_ScheduleDriver()` with the stored handle.

leftink, 2019-08-23:
cocotb does not "drive" signals.  It does deposit, and I believe there is work to do freeze/unfreeze.  This is the same for vhpi and vpi interfaces.

This does have it's limitations, i.e. if a signal has a driver, the only way to override it is with freeze/unfreeze.

Why do you think you need a driver on the signal?

higgns, 2019-08-24:
Tri state busses would be such an example (check out  #1059).

Imo driving signals would be expected testbench behaviour (especially when coming from native VHDL testbenches), but I see that both ways have their use cases.

ktbarrett, 2020-07-10:
xref #1855

closed 2020-10-26 by ktbarrett