cocotb/cocotb #1087: Can cocotb use C++11?

eric-wieser, 2019-08-24:
Might be good to have a minimum supported C++ standard version stated in our docs, just like we have a minimum supported python version.

imphil, 2019-08-26:
More useful than language features is typically specifying the minimum supported compiler, *and* having a demonstrated need for a given feature from a language version.

Probably the most common Linux distribution in the EDA industry is RHEL, next to SLES and Ubuntu.

- RHEL7: GCC 4.8 (https://access.redhat.com/solutions/19458)
- RHEL6: GCC 4.4 (https://access.redhat.com/solutions/19458)

- Ubuntu 16.04: GCC 5.3.1
- Ubuntu 18.04: GCC 7.4.0

- SLES 12 SP3: GCC 4.8
(Documentation for SLES is as usual harder to find than for other distributions.)

For C++11, many things are already available in GCC 4.4, some are not ([docs](https://gcc.gnu.org/projects/cxx-status.html)). Would you be ok with specifying GCC 4.4 as minimum version for now, or do you have a specific feature you'd like to use which isn't available there?

TC01, 2019-08-30:
Note that RHEL6 ships with python 2.6, which is too old to support cocotb anyway. To get python 2.7, you need to either build it yourself or enable a SCL. (I'm currently using cocotb on a RHEL6 server and have done exactly that).

So there is a potential argument to make for moving to gcc 4.8 instead of 4.4, if necessary: it already doesn't work out of the box due to the system Python version.

themperek, 2019-08-30:
@TC01 You can use cond and install locally (https://conda.io/en/latest/)

TC01, 2019-08-30:
Sure-- I would count that as "build it yourself" though. You're not getting the system Python.

It's also possible to get a newer gcc on RHEL6 (again, either by enabling a SCL or setting it up yourself).

PR #1108 (2019-09-14): Docs: Require at least GCC 4.8.1
GCC 4.8.1 is the first version to fully support C++11, and is available
in RHEL7, SLES12SP3, and Ubuntu 16.04 LTS (where GCC is at 5.3.1).

On RHEL6, users already need to add a SCL to get a supported Python
version. They can do the same to get a supported GCC version.

This change allows us to use C++11 in our C++ code.

Fixes #1087
commit 09348db5 (2019-09-14): Docs: Require at least GCC 4.8.1

GCC 4.8.1 is the first version to fully support C++11, and is available
in RHEL7, SLES12SP3, and Ubuntu 16.04 LTS (where GCC is at 5.3.1).

On RHEL6, users already need to add a SCL to get a supported Python
version. They can do the same to get a supported GCC version.

This change allows us to use C++11 in our C++ code.

Fixes #1087
closed 2019-09-14 by imphil