cocotb/cocotb #1089: How to pass args to python module? Got an error in test_plusargs

BLu85, 2019-08-26:
How can I pass the python module a specific file or a set of arguments in order to run a specific test case?

in _cocotb/tests/test_cases/test_plusargs_ I ran:

`make TOPLEVEL_LANG=vhdl SIM=ghdl`

but I got this error: 

`/usr/local/bin/ghdl -r   --work=work tb_top --vpi=/home/xyz/cocotb/tests/test_cases/test_plusargs/build/libs/x86_64/libvpi.so  +foo=bar +test1 +test2 +options=fubar
/usr/local/bin/ghdl:error: unknown option '+foo=bar', try --help`

Thanks,
Luca

themperek, 2019-08-26:
Do you mean this?
https://cocotb.readthedocs.io/en/stable/building.html?highlight=TESTCASE#envvar-TESTCASE

BLu85, 2019-08-26:
I am not sure how to use that. But the test I pointed in the topic is not working using _ghdl, vhdl_
That test shows how to use _cocotb.plusargs_ but it returns an error. 
I would like to run a command like

`make TOPLEVEL_LANG=vhdl SIM=ghdl SEED=123456789 READFILE=a.txt ARG1=x ARG2=y ARG3=z`

where `READFILE`, `ARG1`, `ARG2` and `ARG3` are read by the function decorated in _cocotb.test_

Is `TESTCASE` the way to achieve that? How should I use it?

Thanks

imphil, 2019-08-26:
In this case, just set READFILE=a.txt ARG1=x ARG2=y ARG3=z as normal environment variables, and use os.environ in Python to read it (https://docs.python.org/3/library/os.html#os.environ).

BLu85, 2019-08-26:
> In this case, just set READFILE=a.txt ARG1=x ARG2=y ARG3=z as normal environment variables, and use os.environ in Python to read it (https://docs.python.org/3/library/os.html#os.environ).

That worked.
Many thanks

imphil, 2019-08-26:
Great, thanks for confirming!

themperek, 2019-08-26:
- I may be wrong but plusargs is a Verilog feature so probably does not work with ghdl.
- If you look for generics you can look here: https://github.com/cocotb/cocotb/issues/756
- for seed look: https://cocotb.readthedocs.io/en/latest/building.html#envvar-RANDOM_SEED

themperek, 2019-08-26:
I close the issue but feel free to reopen if needed.