cocotb/cocotb #1101: Improve scheduler error messages

pwiecha, 2019-09-05:
This is an issue for discussion how error messages can be improved, continuation from gitter:

Doing this:
```
@cocotb.test()
def some_test(dut):
    # cocotb.fork(Clock(dut.clk, 800000, units='ns').start())
    yield RisingEdge(dut.clk)
```
gives

> Failing test at simulator request before test run completion: Simulator shutdown prematurely

Can this error message be improved? I've just spend some time debugging a tb only to find out I forgot to create one of my clocks and the message wasn't very helpful ; )

Responses so far:
@garmin-mjames 
>  In this specific case we could add something to the scheduler to list the Triggers in _trigger2coros along with the coroutines waiting on them for debugging

@eric-wieser 
> That's not a bad idea
Slightly more advanced would be to fire all remaining triggers with an exception
Which would allow tests to do cleanup like close open files

garmin-mjames, 2019-09-05:
If we just print out the `Scheduler._trigger2coros` contents, I think the best place would be between these two lines in `RegressionManager.handle_result()`:
https://github.com/cocotb/cocotb/blob/dc4abc4609cf10c55467f9d5caa65c4e0df72df0/cocotb/regression.py#L303-L304

ktbarrett, 2020-01-03:
Is there anyway to determine if the simulator decided to shutdown because of deadlock vs some other issue?

garmin-mjames, 2020-01-03:
As far as I can see, `embed_sim_event()` is only called in two conditions:

1. `embed_sim_init()` fails.
2. Simulator shutdown callback is called (meaning the simulation finished while simulator has control rather than cocotb telling simulator to end)

Condition 1 also currently uses the same error message as 2.

ktbarrett, 2020-01-03:
@garmin-mjames But can you really say anything about what 2 is telling you? It could be deadlock, or it could be an assertion failing. I think we should throw an exception in every living coroutine like what Eric said, but I don't know if we can ever decide if it's deadlock, so we shouldn't always print additional diagnostics. That might be a feature of the scheduler that we can offer the user on demand, for example, when handling a `SimFailure`.

garmin-mjames, 2020-01-03:
Missed one:
3. `gpi_embed_event()` can be called from VPI code.

closed 2025-11-05 by ktbarrett