cocotb/cocotb #1102: VCS Timescale Mismatch?

THofstee, 2019-09-08:
I have a design that has a clock with a period of 10 (in default cocotb units), and a simulation that runs for 85081 cycles. What I see in the report for VCS is the following:

```
*********************************************************************
** TEST         PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
*********************************************************************
** tb.test_app    PASS     85081000000000.00        [redacted]     **
*********************************************************************
```

Well, that's not right, there are 9 extra 0s. 1ns is 1e-9s, so that's a bit suspicious. What does Incisive say?

```
*********************************************************************
** TEST         PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
*********************************************************************
** tb.test_app    PASS        85081.00        [redacted]           **
*********************************************************************
```

Ok, that looks correct.

What's going on here? Did I misconfigure something or is there something wrong with how cocotb is setting up the VCS testbench?

THofstee, 2019-09-08:
I should add that if I query for the `get_sim_time()` it returns 85081 in both cases.

cmarqu, 2019-09-08:
Instead of trying to find out and document what the different simulators have as a default timestep, maybe we should just output a warning telling the users to always include the ``units`` argument?

THofstee, 2019-09-08:
That's a reasonable suggestion, but that raises another issue of setting `'ns'` segfaults VCS with the default makefile.

```
Send raised exception:
Traceback (most recent call last):
  File "cocotb/decorators.py", line 259, in _advance
    return outcome.send(self._coro)
  File "cocotb/outcomes.py", line 37, in send
    return gen.send(self.value)
  File "tb.py", line 182, in test_app
    cocotb.fork(Clock(dut.clk, CLK_PERIOD, 'ns').start())
  File "cocotb/clock.py", line 76, in __init__
    self.period = get_sim_steps(period, units)
  File "cocotb/utils.py", line 122, in get_sim_steps
    time, units, _LOG_SIM_PRECISION))
ValueError: Unable to accurately represent 10(ns) with the simulator precision of 1e0

```

THofstee, 2019-09-08:
Seeing as how cocotb already defaults to reporting everything in NS, maybe adding an overrideable `TIMESCALE` flag to the makefiles that defaults to `1ns/1ns` might not be unreasonable? Most of the other compilers I've tested default to `1ns/1ns`, and VCS even says `` `timescale`` when used defaults to 1ns (though for some ungodly reason they make it default to 1s/1s when not specified anywhere).

cmarqu, 2019-09-09:
That may be the best way forward.
This changes default behavior in a backward-incompatible way for some cases, but these cases were half-broken anyway. 
We are moving toward prefixing our variables with ``COCOTB_``, so ``COCOTB_HDL_TIMESCALE`` sounds good?

eric-wieser, 2019-09-09:
> it seems like cocotb swallowed up the first part of that presumably multiline info and didn't report it.

That's just how python shows tracebacks. You're not losing any relevant information there anyway, although the message could be clearer (ie, using `1s` instead of `1e0`)

THofstee, 2019-09-09:
@cmarqu that sounds reasonable to me. I'm of the opinion that tools should provide predictable and sane defaults, having the defaults change between cocotb backends would be a bit odd, so I think breaking the backwards compatibility is worth it here. Likewise, I don't think 1s/1s is a sane default. cocotb is great because it means I can get predictable behavior across multiple tools and I've been quite pleased with it so far!

In VCS the timescale would be `-timescale=$(COCOTB_HDL_TIMESCALE)` and placed in the compile phase. In the Cadence tools it's `-timescale $(COCOTB_HDL_TIMESCALE)` and placing it in the SIM_ARGS was what I did before but maybe it should be in COMPILE_ARGS. 

@eric_wieser I think I was using whatever was on PyPI. I can try again with master when I get a chance.

cmarqu, 2019-09-09:
Actually, I think we need two environment variables ``COCOTB_HDL_TIMESTEP`` and ``COCOTB_HDL_TIMEPRECISION`` so that we don't rely on the format of the "step/precision" string.

cmarqu, 2019-09-09:
We should have a good regression test here and handle #289 and #405 at the same time.

imphil, 2019-09-14:
Thanks for reporting this bug @THofstee. I agree that having consistent behavior across simulators is a desirable goal, and making sure we choose a suitable default and consistent timescale is a good thing.

So the steps to get this implemented would be:
- Introduce a COCOTB_HDL_TIMESTEP and COCOTB_HDL_TIMEPRECISION makefile/environment variable. Set these variables to sane defaults: 1ns/1ps or 1ns/1ns (probably the former?) Ideally we use something that most simulators already use as default, with VCS being an exception apparently.
- Go through all simulator-specific makefiles and pass this information to it in a simulator-specific fashion.
- Document the new variables at https://cocotb.readthedocs.io/en/latest/building.html#make-variables

@THofstee is this something you could provide a patch for? I'm happy to help out with any questions you might have along the way.

THofstee, 2019-09-16:
The defaults here might not be 100% accurate, they are a combination of what I've experienced and what I've seen other people post because I don't have all the manuals for the various simulators.

| Simulator | Default | Command |
|-----------|---------|---------|
| aldec     | ?       | ? |
| cvc       | ?       | ? |
| ghdl      | ?       | VHDL seems to handle time differently. |
| icarus    | 1s/1s   | Not possible from command line. |
| ius       | 1ns/?   | `-timescale $(COCOTB_HDL_TIMESCALE)/$(COCOTB_HDL_TIMEPRECISION)` |
| modelsim  | 1ns/?   | `-timescale $(COCOTB_HDL_TIMESCALE)/$(COCOTB_HDL_TIMEPRECISION)` |
| nvc       | ?       | VHDL seems to handle time differently. |
| questa    | 1ns/?   | `-timescale $(COCOTB_HDL_TIMESCALE)/$(COCOTB_HDL_TIMEPRECISION)` |
| vcs       | 1s/1s   | `-timescale=$(COCOTB_HDL_TIMESCALE)/$(COCOTB_HDL_TIMEPRECISION)` |
| xcelium   | 1ns/?   | `-timescale $(COCOTB_HDL_TIMESCALE)/$(COCOTB_HDL_TIMEPRECISION)` |

THofstee, 2019-09-16:
Following on the threads @cmarqu mentioned, here are what I think the additional quality of life changes would be:

- Warn if trying to create a clock or schedule a timer that requires smaller precision than has been set by COCOTB_HDL_TIMEPRECISION or the precision that the simulator reports.
- Make the clock module function as a clock generator and adjust its period based on accumulated error.
- Add tests that cover all of the above behavior.

Other misc questions:
- [Why does `self.frequency` in clock.py set units to `'us'`?](https://github.com/cocotb/cocotb/blob/master/cocotb/clock.py#L78) It seems like this might have been the result of updating old code in PR #458, should it be using the units passed in instead?

eric-wieser, 2019-09-16:
It looks like `Clock.frequency` is always measured in kHz, so that conversion is more correct right now than it would be if you forwarded the units

eric-wieser, 2019-09-16:
> Warn if trying to create a clock or schedule a timer that requires smaller precision than has been set by COCOTB_HDL_TIMEPRECISION or the precision that the simulator reports.

I think we already throw an exception if the timer is finer than the simulator resolution?

garmin-mjames, 2019-09-18:
Aldec:
Argument to asim/vsim command

-t <resolution> 
Specifies the simulation resolution. The resolution should be specified as a combination of a number and a time unit: fs, ps, ns, us, ms. If the -t argument is omitted, the resolution is determined automatically. In Verilog designs the maximum time precision specified with the `timescale directive is used. In VHDL designs and Verilog designs without any `timescale directives the resolution defaults to 1 ps.

THofstee, 2019-09-18:
@garmin-mjames is there a similar command for the timestep (the top part of 1ns/1ps)?

garmin-mjames, 2019-09-18:
I'm not sure what exactly the timestep is referring to.
yielding a Timer(1) will yield the same amount of time as the resolution in Aldec.
So if we use `asim -t 100ps` and the first thing we do in the test is `yield Timer(1)`, the simulation will advance 100ps and `get_sim_time()` will return `1`, while `get_sim_time(units='ps')` will return `100.0`.

THofstee, 2019-09-18:
@garmin-mjames If I have a timescale of 1ns/1ps, that means that `#1` in Verilog will delay for 1ns, but the simulator takes steps at a 1ps resolution. So you can make a `#1.001` and it still works. If your resolution is 1ns/1ns, then `#1` and `#1.001` both resolve to `#1`.

The issue I was facing was that the step was not specified so it defaults to 1s in VCS. I don't think cocotb relies on the former of those that much? But if you were using a cocotb testbench with something that had `#delays` in the Verilog, things might go awry quickly if you didn't specify the timestep. If someone specified TIMESTEP=1ns and TIMEPRECISION=1ps, but Aldec assumed step was 1s because the -t flag seems to only set the precision, then I think people would get into strange errors, so it would be nice to specify both.

krupan, 2020-01-19:
I'm running with icarus and seeing this error:
```
ValueError: Unable to accurately represent 50(ns) with the simulator precision of 1e0
```
The table here: https://github.com/cocotb/cocotb/issues/1102#issuecomment-531638033 says that we can't set the timeunit and timeprecision for icarus on the command-line, and https://github.com/cocotb/cocotb/pull/1113 doesn't appear to try to do anything with icarus.  Are `COCOTB_HDL_TIMESTEP` and `COCOTB_HDL_TIMEPRECISION` not supported for icarus?

hofstee, 2020-01-19:
That's correct. I'm not familiar enough with how Icarus works to have added support for it when I made my pull request, and searching for documentation and looking through the code didn't seem to suggest a way to do so. I might have missed something, if you know how to get the expected behaviour it would be great to get a pull request!

krupan, 2020-01-19:
Until we figure out (or fix) Icarus, would it be possible to update the
cocotb documentation for 1.3 to explain that this feature doesn't work for
Icarus?  That would have saved me some time and trouble.  I can look into
creating a documentation patch myself if I know the project is open to that.

cmarqu, 2020-01-21:
@krupan Sure, please do a PR. Not sure about retroactively modifying 1.3, but documentation points to Latest by default anyway which is git master.

garmin-mjames, 2020-01-21:
Aldec update:

For Verilog source files, the `alog` command (Verilog/SystemVerilog compiler) uses the `-timescale <timeunit>/<timeprecision>` argument that applies to modules that do not have the timescale set in the source code file.

cmarqu, 2020-02-08:
> Aldec update:
> 
> For Verilog source files, the `alog` command (Verilog/SystemVerilog compiler) uses the `-timescale <timeunit>/<timeprecision>` argument that applies to modules that do not have the timescale set in the source code file.

@hofstee Would you consider opening another PR that implements this information?

cmarqu, 2020-04-13:
Looks like everything brought up here is done; thanks again for your work @hofstee.

closed 2020-04-13 by cmarqu