cocotb/cocotb #1103: Support VHDL with VCS

cmarqu, 2019-09-08:
Right now, we are using a two-step setup for Synopsys VCS which according to http://www.vlsiip.com/vcs/ is limited to Verilog only. VCS(-MX) is capable of simulating VHDL too with a three-step process using ``vhdlan``.

cmarqu, 2019-10-23:
I'll work on this soon.

cmarqu, 2019-10-24:
I can successfully compile and elaborate ``test_iteration_vhdl``with https://github.com/cmarqu/cocotb/tree/support-vhdl-in-vcs and also bring up the simulator, but it turns out that the ``vhpi_user.h`` that comes with Synopsys VCS-MX is very different from what is in the appendix of the VHDL standard - enums instead of single constants, and struct elements of different types.
(https://github.com/jamiekang/embedded-sw-dev-tools/blob/master/binsim/vhpi_user.h seems to be an older version of it.)

hashhsah, 2019-12-22:
@cmarqu below is an incomplete attempt at mixed verilog/vhdl simulation, with verilog as top-level module. Simulation runs fine, but I didn't try to access anything within the module written in VHDL.

```
ifeq ($(ARCH),x86_64)
EXTRA_ARGS += -full64
endif
VLOGAN_ARGS += -sverilog +define+COCOTB_SIM=1

$(SIM_BUILD)/pli.tab :
	[ -d $(SIM_BUILD) ] || mkdir $(SIM_BUILD)
	echo "acc+=rw,wn:*" > $@

$(SIM_BUILD)/flist_vlog.f : $(VERILOG_SOURCES)
	[ -d $(SIM_BUILD) ] || mkdir $(SIM_BUILD)
	rm -f $@ ; touch  $@
	for f in $(VERILOG_SOURCES); do echo "$$f" >> $@; done

$(SIM_BUILD)/flist_vhdl.f : $(VHDL_SOURCES)
	[ -d $(SIM_BUILD) ] || mkdir $(SIM_BUILD)
	rm -f $@ ; touch  $@
	for f in $(VHDL_SOURCES); do echo "$$f" >> $@; done

# Compilation phase
$(SIM_BUILD)/_vmake.vlogan: $(SIM_BUILD)/flist_vlog.f
	cd $(SIM_BUILD) && \
    source $(SNPS_ENV) && \
	if [ -s flist_vlog.f ]; then \
	   vlogan $(PLUSARGS) $(VLOGAN_ARGS) $(EXTRA_ARGS) -l vlog.log -f flist_vlog.f ; \
	fi && touch _vmake.vlogan

$(SIM_BUILD)/_vmake.vhdlan: $(SIM_BUILD)/flist_vhdl.f
	cd $(SIM_BUILD) && \
    source $(SNPS_ENV) && \
	if [ -s flist_vhdl.f ]; then \
	   vhdlan $(PLUSARGS) $(VHDLAN_ARGS) $(EXTRA_ARGS) -l vhdl.log -f flist_vhdl.f ; \
	fi && touch _vmake.vhdlan

# Linking phase
$(SIM_BUILD)/simv: $(SIM_BUILD)/pli.tab $(COCOTB_LIBS) $(COCOTB_VPI_LIB) $(CUSTOM_COMPILE_DEPS) $(SIM_BUILD)/_vmake.vlogan $(SIM_BUILD)/_vmake.vhdlan
	cd $(SIM_BUILD) && \
    source $(SNPS_ENV) && \
    LD_LIBRARY_PATH=$(LIB_DIR):$(LD_LIBRARY_PATH) TOPLEVEL=$(TOPLEVEL) \
	vcs    $(PLUSARGS) -noIncrComp +lint=TFIPC,PCWM \
	    +acc+1 +vpi \
	    -P pli.tab -load libvpi.so $(COMPILE_ARGS) $(EXTRA_ARGS) \
	    -CFLAGS "-DVCS" \
	    +vcsd -debug_all \
	   	-o simv -l compile.log -top $(TOPLEVEL)
# for future reference, we may use the following options with -use_vpiobj
#	    -use_vpiobj $(COCOTB_SHARE_DIR)/lib/vpi/VpiImpl.cpp \
#	    -CFLAGS "-DVCS -I$(COCOTB_SHARE_DIR)/include" \
#	    -LDFLAGS "-L$(LIB_DIR) -lvpi -lgpi -lgpilog" \

# Execution phase
results.xml: $(SIM_BUILD)/simv $(PYTHON_FILES) $(CUSTOM_SIM_DEPS)
	-source $(SNPS_ENV) && \
	PYTHONPATH=$(LIB_DIR):$(PWD):$(PYTHONPATH) LD_LIBRARY_PATH=$(LIB_DIR):$(LD_LIBRARY_PATH) \
	    MODULE=$(MODULE) TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) COCOTB_SIM=1 \
		TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
        $(SIM_BUILD)/simv +define+COCOTB_SIM=1 $(SIM_ARGS) $(EXTRA_ARGS)

.PHONY: clean
clean::
	-@rm -rf $(SIM_BUILD)
	-@rm -rf results.xml
	-@rm -rf *.fsdb
	-@rm -rf *.log
	-@rm -rf ucli.key
```

cmarqu, 2019-12-22:
@hashhsah Thanks. This is probably working because VPI is used to due to the toplevel being Verilog, so the VHPI problem I encountered does not show up.
It's still a nice improvement over what we have. (I have to dig up my changes to ``Makefile.vcs`` and compare with your options.)
I'd be interested to hear whether you can discover and access things inside the VHDL part etc.

Blebowski, 2021-07-02:
Hello,
could I please ask what is the state of VHPI support for VCS?

themperek, 2021-07-02:
As I remember last time we looked VCS used different header than our https://github.com/cocotb/cocotb/blob/master/cocotb/share/include/vhpi_user.h and bit no idea where it is coming from. 

We could to the synopsis version on setup. Maybe this would work.

DiegoRosales, 2022-05-23:
Hi!
I'm new to Cocotb and I see that the last update was last year. Does anybody know if there's been any progress with this?

imphil, 2022-05-23:
Hi @DiegoRosales. AFAIK, there was no progress on this issue so far. We do not have VCS available for testing, and no cocotb user with VCS has helped out so far. If you'd like to use VCS with cocotb you're invited to join the development and open a pull request. Please reach out if there's anything we can help you with.

lfanals-slac, 2024-08-27:
Hi, I have access to `vcs`. I have a simple working example with the following structure:
```
.
├── hdl
│   ├── counter.vhdl
│   ├── counter_wrapper.vhdl
│   └── toplevel.sv
├── Makefile
└── tb.py
```
Similarly to @hashhsah, I use a SystemVerilog top level (`toplevel.sv`) and the simulation runs fine. I observe that only the signals in such module can be accessed. I can only read the signals in both vhdl modules via out of module referencing (e.g., `assign oomr = i_counter_wrapper.i_counter.count;`), which is not ideal.

Please advise what's the best way to proceed in order to solve this issue.

themperek, 2025-01-02:
VCS VHPI API (looks like this https://github.com/Blebowski/iso-16845-compliance-tests/blob/master/src/cosimulation/vcs_vhpi_user.h
) is quite different than one we are using (https://github.com/cocotb/cocotb/blob/master/src/cocotb/_vendor/vhpi/vhpi_user.h).

One would need to adopt https://github.com/cocotb/cocotb/tree/master/src/cocotb/share/lib/vhpi

BTW: The standard can be found here  https://standards.ieee.org/wp-content/uploads/import/download/1076-2008_downloads.zip