cocotb/cocotb #1105: Cannot access signal in hierarchy in mixed-language simulation

svancau, 2019-09-10:
Hi,

I'm working on a VHDL design having a Verilog model in it. I get the following error :
> Send raised exception: i_ADC_ip contains no object named modular_adc_0

Where i_ADC_ip is a VHDL module and modular_adc_0 is a Verilog one. I'm using Modelsim 10.6d under Windows 10.

I infer it managed to traverse all the VHDL hierarchy till i_ADC_ip.
The path in Modelsim is  
> sim:/myproject_toplevel/i_functional/i_analog_in/i_ADC_ip/modular_adc_0/control_internal/adc_inst

I accessed it using 
> dut.i_functional.i_analog_in.i_ADC_ip.modular_adc_0.control_internal.adc_inst

How can I fix this issue ?
Thanks

cmarqu, 2019-09-10:
Related: #1004.
Does it help to "discover" everything beforehand as described in the end of that ticket?

garmin-mjames, 2019-09-10:
I'm assuming you're using VHPI for top-level access.
Do you have environment variable `GPI_EXTRA=vpi` to load the VPI library?