cocotb/cocotb #1111: [docs] Document usage of assert

imphil, 2019-09-15:
Since #1110, `assert`s are equal to `raise TestFailure`. We should document this, and possibly even advocate using `assert`.

eric-wieser, 2019-09-16:
Strictly they're not identical, since the latter is considered a `TestComplete`, which currently results in slightly different logs for forked coroutines. #922 might make these mean different things.

ktbarrett, 2019-09-16:
@eric-wieser This commit more or less makes `AssertionError`s a type of expected test end which are handled correctly. It should just be added to the exception list.

The reason there is split handling is because there are some exceptions that when raised in a forked coroutine should cause immediate test failure, and you have to segregate them from the types of errors that are meant to be caught later by the user when yielding on a coroutine join.

eric-wieser, 2019-09-17:
> there are some exceptions that when raised in a forked coroutine should cause immediate test failure

This is true for compatibility reasons, but I remain unconvinced whether it's good design. If we decide it's bad design, the `assert` / `raise TestFailure` split is a great place to change it without breaking compatibility. If we decide it's good design, then yes, making them equivalent would be sensible.

ktbarrett, 2019-09-18:
> I remain unconvinced whether it's good design

Using `TestComplete` is not a good design, but allowing the user to fail-hard or finish immediately in a forked coroutine is a required behavior. Currently, `TestComplete` is the most reasonable implementation seeing as we are still using it throughout the scheduler and regression manager, and it's how users are writing their tests.

I really dislike how exceptions are handled, I've complained about a different aspect of exception handling before in #936. Personally, I like the idea of the scheduler or regression manager having a functional interface for stopping tests instead of abusing exceptions (and it is abuse, it's incomprehensible, see the latest commit for erasing parts of the exception traceback. **Exceptions should not be used for control flow**). For example, `cocotb.panic(msg="")`, `cocotb.fail(msg="")`, `cocotb.pass(msg="")`. We could then offer decorators for coroutines that wraps them in a monitor that translates all caught exceptions to `cocotb.panic(msg=str(exception))`, like `fork`s current behavior, or it could just log the exception or outcome, etc... By default, a `fork` like function (I like `run`) should simply launch a coroutine and catch any exception thrown in that coroutine as an outcome. I went over some of this in #937. 

If you mean you need an argument on why fail-hard or finish immediately are important I can give you pages.

eric-wieser, 2019-09-18:
> Personally, I like the idea of the scheduler or regression manager having a functional interface for stopping tests instead of abusing exceptions (and it is abuse, it's incomprehensible, see the [...]

My motivation here has been to, at least for single-"threaded" tests, follow the established python convention of failing a test via an assertion error (see: pytest, unittest, nose, ...).

Using exceptions for the flow / propagation _of failures_ is a very appropriate use. I can think of one or two places where the scheduler _does_ use them for control flow, but it does so only in a way that mimics how python as a language uses `StopIteration` for control flow. If there's a particularly egregious case I've missed, please draw my attention to it.

By using exceptions instead of a `cocotb.panic`-style API, cleanup becomes theoretically possible via `with` statements.

I'm not sure this is the best place to discuss this, but it's better to discuss it somewhere that nowhere.

ktbarrett, 2019-09-18:
All of this talk got me looking and I'm fairly certain that throwing an `AssertionError` in a forked coroutine will cause a test error instead of a failure. Haven't tested though. Relevant line is 503.

https://github.com/cocotb/cocotb/blob/ce5cc04ea56539a0dddf8fa1492d0b1f525e747a/cocotb/scheduler.py#L499-L510

ktbarrett, 2022-03-09:
Closed by #2492.

closed 2022-03-09 by ktbarrett