cocotb/cocotb #1115: Timer value argument is misleading

cmarqu, 2019-09-18:
With the existence of the ``units`` argument, the value in ``Timer`` should not be ``time_ps`` (https://cocotb.readthedocs.io/en/latest/triggers.html?highlight=Timer#cocotb.triggers.Timer). 

How about ``value``?

ktbarrett, 2019-09-20:
Regarding this and #1117. 

I don't even think the default units are picoseconds, but the simulator's timescale, which can change via flags to the simulator. Not using `units` is probably a bug/bug waiting to happen. Maybe a `DeprecationWarning` is appropriate.

cmarqu, 2019-09-20:
Maybe we should even take the step and create ``SimTime`` as @imphil suggested in #872.

ktbarrett, 2019-09-26:
I'd like something like `std::chrono::duration` for expressing sim time, I've run into the need for such a type on a few occassions.

ktbarrett, 2020-12-12:
@garmin-mjames What is the request for 1.5 here? The rename, or the simulation time physical type I described?

garmin-mjames, 2020-12-12:
A rename was my goal for 1.5.
Supporting a physical `time` type in Python would help #1233 as well.

cmarqu, 2021-01-14:
https://stackoverflow.com/questions/49802412/how-to-implement-deprecation-in-python-with-argument-alias could be useful here

ktbarrett, 2021-01-14:
The first part of the first solution would not be good for Sphinx, but the second part would work fine. I would introduce two special objects that would be helpful for showing the intended use.

```python
DEPRECATED = object()
REQUIRED = object()

def function(new_name=REQUIRED, *, old_name=DEPRECATED):
    # ... logic from answer 1 part 2 just replacing `None`, `None` might be a possible argument anyway, so we need a better sentinel
```

PR #2352 (2021-01-19): Timer param deprecation
Closes #1115.
closed 2021-01-19 by ktbarrett