cocotb/cocotb #1126: Cocotb logo

themperek, 2019-09-25:
Together with @cmarqu we made a logo proposition for cocotb. Would be nice to know your opinion and decide yes/now for this one or propose other.

[image]

[image]

imphil, 2019-09-25:
Can you explain the logo a bit? What does the symbol stand for?

themperek, 2019-09-25:
Up to interpretation but for me is square waveform (clock) made out of python logo elements.

tmeissner, 2019-09-25:
I like it. Maybe you could add a waveform in the image to clarify.

eric-wieser, 2019-09-25:
Not a huge plan of the cut and paste approach from the python logo, but I like the concept.

mgielda, 2019-12-03:
Hm, nice idea, but it is indeed a tad controversial with the cut and paste. What if we just had the text (cocotb in this font) for now? That would be a good first step as we discuss the symbol further. Otherwise we keep running into the problem of cocotb not having a logo ;)

themperek, 2019-12-03:
Maybe this will be ok?

[image]

Or this (green wave is chasing red one ?)

[image]

eric-wieser, 2019-12-03:
@themperek, what's the font in the first image?

Also, can you upload these as SVG to make it easier for others to edit and suggest changes to?

themperek, 2019-12-03:
The font is: Neuropolitical

Files: https://uni-bonn.sciebo.de/s/pTbCDZEVyqeDZCr

cmarqu, 2019-12-03:
I like the blue/yellow version for its clear reference to Python that is much less obvious in the green/red one.

eric-wieser, 2019-12-03:
How's this?

[image]

[logo_efw.zip](https://github.com/cocotb/cocotb/files/3916276/logo_efw.zip)

cmarqu, 2019-12-03:
I'd say @themperek's last version has the advantage that it would work even in very small resolutions.

themperek, 2019-12-03:
@eric-wieser We could make a poll https://github.com/apps/polls and wait for some time?

eric-wieser, 2019-12-03:
@themperek: I thought about that exact tool, but it doesn't allow options to be added later - so perhaps we should wait a little longer for other options.

themperek, 2021-04-05:
[image]

themperek, 2021-04-05:
[image]

wallento, 2021-04-06:
I prefer the upper right I believe

eric-wieser, 2021-04-06:
I'd be tempted to flip them left/right - waveforms usually grow to the right

ktbarrett, 2021-04-06:
I agree with Eric. The top left or top middle but mirrored horizontally would like nice. Over or next to the "cocotb" in the Neuropolitical font as suggested earlier. For my course in lieu of a decision I actually used Eric's contribution.

themperek, 2021-04-06:
[image]

ktbarrett, 2021-04-06:
I think I might like the one without eyes more, but I like both.

themperek, 2021-04-06:
Can look for other/better font. Many free ones: https://www.dafont.com/

eric-wieser, 2021-04-06:
I think the eyes look better when connected to the heads 😉 

[image]

PackDevLab, 2021-04-08:
Hi! Here is a suggestion for new logo. 
The C is representing the testbench an the arrow representing the data flow. 
I have also suggested some different variations in color and finish.
 
[image]
[image]
[image]
[image]

PackDevLab, 2021-04-08:
As a comment to the suggestions above; i believe the python snakes are to close to the original python logo, in all versions. As this is a rather conservative area, i have gone for a less playful style, but still trying to show a bit of creativity.

themperek, 2021-04-08:
@PackDevLab Thank you!

What you think to make the first symbol look more like a flip-flop? Maybe this would give a bit more relation to electronics.

[image]

eric-wieser, 2021-04-08:
> i believe the python snakes are to close to the original python logo, in all versions

For a good overview of how "snakey" other logos for python packages are, https://python3statement.org/#sections30-projects has a large collection of logos across popular packages.

ktbarrett, 2021-04-08:
@PackDevLab 
> The C is representing the testbench an the arrow representing the data flow.

Could you explain that a bit more?

eric-wieser, 2021-04-14:
[image]

A possible variant of @PackDevLab's design, using the "C" to represent the input-output. Very rough

themperek, 2021-04-14:
[image]

PackDevLab, 2021-04-18:
Here are some sketches based on previous comment. I also tried just adding colors as shadows to the "o"s .
I believe making the C look to much like a snake could be a bit too litteral, so my take on the resemblance to Python logo was to separate the blue and yellow in the same style as they do. The sketch doesnt show that correctly due to lack of time, but i thing you get the gist.  I used an "arrow" instead of snake to symbolize the flow.
[image]
[image]
[image]

wallento, 2021-04-18:
For the first one, there could be an "icon" version with the two arrows only then, which I always like.

marlonjames, 2021-04-19:
Playing with the chip idea. I used gold/yellow for the input, as in "golden" stimulus/reference.

[image]

alexforencich, 2021-04-19:
I think we also need a mascot of some sort, perhaps named "coco".

ktbarrett, 2021-04-19:
> perhaps named "coco".

[image].

PackDevLab, 2021-04-19:
I absolutely love the yellow/blue with chip "o",  without an outline. 
I also absolutely love Coco but that might be to intrusive and also associating to Crash might be contra productive ;)

imphil, 2021-04-19:
@marlonjames The arrow and the chip look good to me. The font I don't like at all honestly, it reminds me of a free font that tries to look playful, but just doesn't get the balance between rounded and rectangular right.

eric-wieser, 2021-04-19:
I think the arrow looks much better the way @marlonjames draws it with the endpoints rather than the center aligning with the center of the "o" - the way I had it makes it harder to read as a "c"

marlonjames, 2021-04-19:
@imphil I think any nice font with a "rounded rectangle" style would work.
For the arrows I started with a larger sized capital `C` so the rounded corners matched.

marlonjames, 2021-04-19:
Are any of these appealing?
https://www.fontsc.com/font/tag/rounded-square?pt=Cocotb&ps=2&o=1&ls=50&co=1

eric-wieser, 2021-04-19:
I quite like "Anita semi-square" (ignoring the odd kerning the `tb` looks interesting) and "Play" (the o looks somewhat chip-shaped already).

marlonjames, 2021-04-19:
Here's with the **Play** font.

[image]

imphil, 2021-04-24:
Cocotb now has a logo, which will be revealed as part of a web site launch in the coming months. Thank you all for your input!

closed 2021-04-24 by imphil