cocotb/cocotb #1131: Conda Environment "Did not detect Python"

dan-fritchman, 2019-09-28:
It seems that cocotb is tripped up by finding the Python interpreter in a Conda environment. 

To reproduce: 
* Create a fresh conda env
* Activate said conda env
* Follow the [Quickstart](https://github.com/cocotb/cocotb#quickstart) instructions 

Commands:

```
conda create --name cocotb-py37-env python=3.7
conda activate cocotb-py37-env
git clone https://github.com/cocotb/cocotb
pip install ./cocotb
cd cocotb/examples/endian_swapper/tests
make
```

Generates these errors:

```
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
```

themperek, 2020-01-22:
@dan-fritchman This should not happen and I cannot reproduce this. Can you try with the latest version?

ktbarrett, 2020-07-10:
This should be resolved now, if it is not, please re-open.

closed 2020-07-10 by ktbarrett