cocotb/cocotb #1133: Use of a cocotb signal in expression does not work

svancau, 2019-09-30:
Hi,

During my tests I found out constructs like this :
`mysignal <= 0 if myinputsignal == 3 else 1`

typically don't work and just don't affect outputs. Whereas 

```
if myinputsignal == 3:
  mysignal <= 0
else:
  mysignal <= 1

```
just works

Do you have an idea why it doesn't work ?

eric-wieser, 2019-09-30:
Operator precedence is not what you think it is - this probably should go in the docs. You wrote:

```
mysignal <= 0 if myinputsignal == 3 else 1
```
Which means
```
(mysignal <= 0) if myinputsignal == 3 else 1
```
But you meant
```
mysignal <= (0 if myinputsignal == 3 else 1)
```

ktbarrett, 2019-09-30:
I guess it's a little late to start complaining about how `<=` was a bad idea. It precludes the usage of signals in comparisons. You can overload assignment by modifying the ast prior to loading cocotb, https://github.com/RyanKung/assign.

svancau, 2019-10-01:
Thanks, I just didn't figure out it was an operator precedence issue.

closed 2019-10-01 by svancau