cocotb/cocotb #1148: Transitive includes in C/C++ sources

ktbarrett, 2019-10-14:
This is bad practice. Discovered after trying to use clang-format on the source. Clang-format reordered the includes and cocotb stopped compiling.

eric-wieser, 2019-10-14:
Do you have any recommendations for how to find and fix these?

garmin-mjames, 2019-10-14:
Something like https://include-what-you-use.org/ might help here.

garmin-mjames, 2020-01-09:
I ran **IWYU** on the tip of your #1137 branch.

> Environment:
> - Windows Subsystem for Linux (WSL) Ubuntu 18.04
> - `include-what-you-use 0.8 (git:c6c4520) based on clang version 4.0.0 (tags/RELEASE_400/final)`

Here's the output for `cpp` files:
```
.../include-what-you-use/bin/include-what-you-use -fno-common -g -DDEBUG -fpic -std=c++11 -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -DICARUS -Wall -Wextra -Wconversion -Wcast-qual -Wwrite-strings -Werror -Wnon-virtual-dtor -Woverloaded-virtual  -c -I/tmp/cocotb-clang/cocotb/share/include -I/usr/include/python2.7  -o /tmp/cocotb-clang/build/obj/x86_64/GpiCbHdl.o GpiCbHdl.cpp

GpiCbHdl.cpp should add these lines:
#include <stdio.h>        // for snprintf
#include <stdlib.h>       // for malloc, NULL, size_t
#include <string.h>       // for strlen
#include <string>         // for string, operator==
#include "gpi.h"          // for gpi_edge::GPI_FALLING, gpi_edge::GPI_RISING
#include "gpi_logging.h"  // for LOG_WARN, LOG_DEBUG, LOG_CRITICAL, LOG_ERROR

GpiCbHdl.cpp should remove these lines:

The full include-list for GpiCbHdl.cpp:
#include <stdio.h>        // for snprintf
#include <stdlib.h>       // for malloc, NULL, size_t
#include <string.h>       // for strlen
#include <string>         // for string, operator==
#include "gpi.h"          // for gpi_edge::GPI_FALLING, gpi_edge::GPI_RISING
#include "gpi_logging.h"  // for LOG_WARN, LOG_DEBUG, LOG_CRITICAL, LOG_ERROR
#include "gpi_priv.h"     // for GpiCbHdl, GpiValueCbHdl, GpiObjHdl, GpiHdl
---
.../include-what-you-use/bin/include-what-you-use -fno-common -g -DDEBUG -fpic -std=c++11 -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -DICARUS -Wall -Wextra -Wconversion -Wcast-qual -Wwrite-strings -Werror -Wnon-virtual-dtor -Woverloaded-virtual  -c -I/tmp/cocotb-clang/cocotb/share/include -I/usr/include/python2.7  -o /tmp/cocotb-clang/build/obj/x86_64/GpiCommon.o GpiCommon.cpp

GpiCommon.cpp should add these lines:
#include <bits/stdint-intn.h>   // for int32_t
#include <bits/stdint-uintn.h>  // for uint32_t, uint64_t
#include <stdio.h>              // for printf
#include <stdlib.h>             // for NULL, exit, getenv, size_t
#include <string>               // for string, operator+, operator==, basic_...
#include <utility>              // for pair
#include "embed.h"              // for embed_sim_event, embed_init_python
#include "gpi.h"                // for gpi_sim_hdl, gpi_iterator_hdl, gpi_ev...
#include "gpi_logging.h"        // for LOG_DEBUG, LOG_ERROR, LOG_WARN, LOG_INFO

GpiCommon.cpp should remove these lines:
- #include <sys/types.h>  // lines 32-32
- #include <unistd.h>  // lines 33-33

The full include-list for GpiCommon.cpp:
#include <bits/stdint-intn.h>   // for int32_t
#include <bits/stdint-uintn.h>  // for uint32_t, uint64_t
#include <cocotb_utils.h>       // for utils_dyn_open, utils_dyn_sym, xstr
#include <stdio.h>              // for printf
#include <stdlib.h>             // for NULL, exit, getenv, size_t
#include <map>                  // for map, map<>::iterator, _Rb_tree_iterator
#include <string>               // for string, operator+, operator==, basic_...
#include <utility>              // for pair
#include <vector>               // for vector, vector<>::iterator, allocator
#include "embed.h"              // for embed_sim_event, embed_init_python
#include "gpi.h"                // for gpi_sim_hdl, gpi_iterator_hdl, gpi_ev...
#include "gpi_logging.h"        // for LOG_DEBUG, LOG_ERROR, LOG_WARN, LOG_INFO
#include "gpi_priv.h"           // for sim_to_hdl, GpiImplInterface, GpiObjHdl
---
...
.../include-what-you-use/bin/include-what-you-use -fno-common -g -DDEBUG -fpic -std=c++11 -DVPI_CHECKING -DICARUS -Wall -Wextra -Wconversion -Wcast-qual -Wwrite-strings -Werror -Wnon-virtual-dtor -Woverloaded-virtual  -c -I/tmp/cocotb-clang/cocotb/share/include -I/usr/include/python2.7  -o /tmp/cocotb-clang/build/obj/x86_64/VpiImpl.o VpiImpl.cpp

VpiImpl.h should add these lines:
#include <bits/stdint-intn.h>   // for int32_t
#include <bits/stdint-uintn.h>  // for uint32_t, uint64_t
#include <string.h>             // for memset, NULL
#include <string>               // for string
#include "gpi.h"                // for gpi_objtype_t, gpi_edge::GPI_FALLING
[... 47 lines cut ...]
#include <string>               // for basic_string, string, operator+
#include <vector>               // for vector, vector<>::iterator
#include "../gpi/gpi_priv.h"    // for GpiObjHdl, GpiIterator, GpiIteratorMa...
#include "gpi.h"                // for gpi_objtype_e::GPI_GENARRAY, gpi_sim_...
#include "gpi_logging.h"        // for LOG_DEBUG, LOG_CRITICAL, FENTER, LOG_...
#include "sv_vpi_user.h"        // for vpiStructNet, vpiStructVar, vpiIntVar
#include "vpi_user.h"           // for vpiHandle, vpi_get_value, s_vpi_value

VpiCbHdl.cpp should remove these lines:

The full include-list for VpiCbHdl.cpp:
#include <bits/stdint-intn.h>   // for int32_t
#include <bits/stdint-uintn.h>  // for uint32_t, uint64_t
#include <stdio.h>              // for NULL, fprintf, stderr
#include <cstddef>              // for size_t
#include <string>               // for basic_string, string, operator+
#include <vector>               // for vector, vector<>::iterator
#include "../gpi/gpi_priv.h"    // for GpiObjHdl, GpiIterator, GpiIteratorMa...
#include "VpiImpl.h"            // for VpiSignalObjHdl, VpiCbHdl, VpiIterator
#include "gpi.h"                // for gpi_objtype_e::GPI_GENARRAY, gpi_sim_...
#include "gpi_logging.h"        // for LOG_DEBUG, LOG_CRITICAL, FENTER, LOG_...
#include "sv_vpi_user.h"        // for vpiStructNet, vpiStructVar, vpiIntVar
#include "vpi_user.h"           // for vpiHandle, vpi_get_value, s_vpi_value
---
```

eric-wieser, 2020-01-09:
I'm a little confused by `bits/stdint-intn.h` being suggested there, which I did not think was a standard header -  was it trying to recommend `<cstdint>`?

garmin-mjames, 2020-01-09:
I'm not sure, it could be a result of my setup.
I would focus mostly on the `cocotb` includes and making sure each file has the proper includes for things that it uses.
IWYU also doesn't analyze `.c` files, so those will have to be checked manually or with some other tool.

ktbarrett, 2020-01-10:
@garmin-mjames I have include fixes staged on top of #1137 including C file changes. I'm waiting for resolution of #1137 before doing the PR. I personally don't like having a bunch of PRs hanging around.

cmarqu, 2020-07-27:
How does it look these days?

ktbarrett, 2020-07-27:
@cmarqu No better.