cocotb/cocotb #1154: No .vcd trace dump using Icarus.

jurevreca12, 2019-10-17:
Hello,
I am having trouble getting the graphic output of the simulation from Icarus.

This is my testbench:
```
import cocotb                                                                                                                   
from cocotb.triggers import Timer, RisingEdge                                                                                   
from cocotb.clock import Clock                                                                                                  
from cocotb.result import ReturnValue, TestFailure                                                                              
                                                                                                                                
CLK_PERIOD = 1000 # ns or 1 MHz                                                                                                 
                                                                                                                                
@cocotb.coroutine                                                                                                               
def reset_dut(rstn, duration):                                                                                                  
    rstn <= 0                                                                                                                   
    yield Timer(duration)                                                                                                       
    rstn <= 1                                                                                                                   
    cocotb.log.info("Reset complete.")                                                                                          
                                                                                                                                
@cocotb.coroutine                                                                                                               
def read_reg(dut, addr):                                                                                                        
    dut.addr_i <= addr                                                                                                          
    yield RisingEdge(dut.clk_i)                                                                                                 
    raise ReturnValue(dut.data_o.value)                                                                                         
                                                                                                                                
                                                                                                                                
@cocotb.test()                                                                                                                  
def basic_write_read(dut):                                                                                                      
    dut._log.info("Running basic_write_read test!")                                                                             
                                                                                                                                
    # Set inputs to zero                                                                                                        
    dut.clk_i  <= 0                                                                                                             
    dut.rstn_i <= 0                                                                                                             
    dut.addr_i <= 0                                                                                                             
    dut.data_i <= 0                                                                                                             
    dut.we_i   <= 0                                                                                                             
                                                                                                                                
                                                                                                                                
    # Start the clock pin wiggling                                                                                              
    cocotb.fork(Clock(dut.clk_i, CLK_PERIOD).start())                                                                           
                                                                                                                                
    # First reset the block                                                                                                     
    yield reset_dut(dut.rstn_i, 2000)                                                                                           
                                                                                                                                
                                                                                                                                
    # Check all registers are set to zero at first                                                                              
    for i in range (0, 31):                                                                                                     
        reg_val = yield read_reg(dut, i)                                                                                        
        if reg_val != 0:                                                                                                        
            raise TestFailure("ERROR: Register x"+ str(i) +" is non-zero after reset! Its value is " + str(reg_val.value) + ".")
                                                                                                                                
    dut._log.debug("Test basic_write_read finnished.")      
```

and my dut:
```
module jedro_1_regfile
#(
    parameter   DATA_WIDTH = 32,
    parameter   ADDR_WIDTH = $clog2(DATA_WIDTH)
)
(
    input                       clk_i,
    input                       rstn_i,
    input   [ADDR_WIDTH-1:0]    addr_i,
    input   [DATA_WIDTH-1:0]    data_i,
    input                       we_i,
    output reg [DATA_WIDTH-1:0] data_o
);

[... 77 lines cut ...]
     0.00ns INFO     cocotb.jedro_1_regfile                  test_regfile.py:24   in basic_write_read                Running basic_write_read test!
2000000000000.00ns INFO     cocotb                                  test_regfile.py:13   in reset_dut                       Reset complete.
32001000000000.00ns INFO     cocotb.regression                         regression.py:266  in handle_result                   Test Passed: basic_write_read
32001000000000.00ns INFO     cocotb.regression                         regression.py:210  in tear_down                       Passed 1 tests (0 skipped)
32001000000000.00ns INFO     cocotb.regression                         regression.py:375  in _log_test_summary               ***************************************************************************************
                                                                                                                             ** TEST                           PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                             ***************************************************************************************
                                                                                                                             ** test_regfile.basic_write_read    PASS     32001000000000.00          0.03   1105316695659334.75  **
                                                                                                                             ***************************************************************************************
                                                                                                                             
[... cut ...]
                                                                                                                             *************************************************************************************
                                                                                                                             
32001000000000.00ns INFO     cocotb.regression                         regression.py:219  in tear_down                       Shutting down...
make[1]: Leaving directory '/home/jurevreca12/Projekti/riscv-jedro-1/sim'
```

I appritiate any help on this issue,
With regards,
Jure

Martoni, 2019-10-17:
Hi @jurevreca12 ,
I'm using Icarus verilog intensively too. I had the same problem at the begining.

In fact, vcd traces are dumped by Icarus only if it asked in the verilog code. If you this code at the end of your module, you will get the traces: 
```verilog
module button_deb(
        // sync design
        input    clk,
        input    rst,
        // in-out
        input    button_in,
output button_valid);

//... verilog module code here

// the "macro" to dump signals
`ifdef COCOTB_SIM
initial begin
  $dumpfile ("button_deb.vcd");
  $dumpvars (0, button_deb);
  #1;
end
`endif

endmodule
```
This is an example from a simple module that [blink a led](https://github.com/Martoni/blp/blob/master/verilog/src/button_deb.v). It generate a wave file named button_deb.vcd that can be visualized with gtkwave : 

```
$ gtkwave button_deb.vcd
```

Now I mainly write HDL with Chisel that generate the verilog code. Then I wrote a little python script that "cocotbify" the top module to [get the waves](https://github.com/Martoni/cocotbify).

I'm not sure that is the right way to do it, but it works.

themperek, 2019-10-17:
Is it OK to close this now?

This is not Icarus nor Cocotb issues but a Verilog itself in this case. 
PS. Some/Most simulators have extra commands for this too.

cmarqu, 2019-10-17:
We should probably document the ``WAVES`` variable (even though we only support it for Aldec and Mentor right now) and can add a note about Icarus there.

Martoni, 2019-10-18:
Just commited a documentation proposal. I think this issue could be closed.

jurevreca12, 2019-10-18:
@Martoni 
Thank you for your answer. It works now yes. This issue can be closed.

closed 2019-10-18 by themperek