cocotb/cocotb #1158: Setting arbitrary clock phase

matthuszagh, 2019-10-19:
Is there a convenient way to generate a clock with an arbitrary clock phase? From [this pr](https://github.com/cocotb/cocotb/pull/1036) it looks like it's possible to set the clock polarity. Is it possible to generalize this and allow arbitrary phase setting? One example where this would be useful is when simulating clocks synchronized via a PLL.

cmarqu, 2019-10-19:
This should be possible, it just looks like nobody has needed it yet.
While we are at it, setting the duty cycle may also be useful - we need to take care though to not slow down the "simple" use case.

eric-wieser, 2019-10-19:
Note that at a certain point it might just be easier to write your own coroutine, it's really not hard, and it gives you complete freedom to change the frequency and period mid simulation:
```
@cocotb.coroutine
def my_custom_clock():
    yield Timer(my_initial_delay)
    while True:
        sig <= 1
        yield Timer(my_high_delay)
        sig <= 0
        yield Timer(low)
```

cmarqu, 2019-10-19:
@eric-wieser BTW, would it be a bit more efficient to create the two high/low Timer instances outside of the loop and await those inside the loop, or does it not matter?

eric-wieser, 2019-10-19:
You're right, it would

cmarqu, 2019-10-20:
Maybe we should just put that example into the docstring of ``Clock()``.

matthuszagh, 2019-10-20:
Thanks for posting the manual solution, that works for me. I agree it would be nice to have it in the docstring.

eric-wieser, 2019-11-20:
@matthuszagh: This is now in the docstring

PR #1172 (2019-11-20): Add custom clocking example to docstring of Clock().
Closes #1158.
commit 5c256095 (2019-11-20): Add custom clocking example to docstring of Clock() (#1172)

Closes #1158, by providing a recipe for the feature requested there in the docs.
closed 2019-11-20 by eric-wieser