cocotb/cocotb #1161: Sending floating point value with BinaryValue

speq, 2019-10-22:
In vcs and ius, vpi_put_value is not supporting string / real value type.
Work around is sending floating point number with BinaryValue and use $bitstoreal translation in verilog.

example:
in py
     dut.co_real_s = BinaryValue(0.345)
in verilog
     always @(*) co_real = $bitstoreal(co_real_s);

with patch
+        elif isinstance(value, float):
+            self.binstr = ''.join(format(i, '08b') for i in reversed(bytearray(struct.pack('d', value))))

[binary.patch.gz](https://github.com/cocotb/cocotb/files/3754319/binary.patch.gz)

cmarqu, 2019-10-22:
Maybe related: #785

ktbarrett, 2021-01-31:
I don't think that conversion is a universal one and is easily provided as a free function and not as a builtin.

closed 2021-01-31 by ktbarrett