cocotb/cocotb #1163: vpi_remove_cb cause vcs crash on scheduler. use vpi_free_object instead

speq, 2019-10-22:
In VCS, using vpi_remove_cb when Exception raise cause simulator crashed. Below is the simulation report:

    Internal error: Unable to locate Eblk in scheduler
    Assertion failed "0" at line 14935 in file sched.c
    An unexpected termination has occurred in ./simv due to an assertion failure in the tool
    Hostname eda
    
    During vpi_remove_cb Callback Reason="6 [cbReadWriteSynch]"
    During a VPI callback function for callback reason="cbStartOfSimulation"
    
    Command line: ./simv
    
    --- Stack trace follows:
    
    Dumping VCS Annotated Stack:
    #0  0x00007fffefe38a3c in waitpid () from /lib64/libc.so.6
    ...
    #12 0x00007fffeea38347 in VpiCbHdl::cleanup_callback (this=0xf22f70) at VpiCbHdl.cpp:106
    #13 0x00007fffeea353ab in VpiImpl::deregister_callback (this=0xf22f60, gpi_hdl=0xf22fc8) at VpiImpl.cpp:520
    #14 0x00007fffee6144ab in gpi_deregister_callback (hdl=0xf22fc8) at GpiCommon.cpp:613
    #15 0x00007fffe35e3e1b in deregister_callback (self=0x7fffe3e21a90, args=0x7fffe32c8310) at simulatormodule.c:904

I have an exception from python script. It random crashes VCS.

According to discussion on Solvnet [Removing VPI Callback or Setting the Handle Free](https://solvnet.synopsys.com/retrieve/025709.html), 

> Question:
> When to use vpi_free_object or vpi_remove_cb?
> 
> Answer:
> 
> Both vpi_free_object and vpi_remove_cb callbacks improve the run-time performance and reduce memory used by the VPI application. But as stated above, vpi_free_object does not remove the scheduled callbacks, hence the choice actually is to be made based on your scenario.
> 
> In general,
> 
> Use vpi_free_object when the application demands callback to exist for later stages of simulation.
> Use vpi_remove_cb when the application doesn't need to have any more callback in the simulation.

I use vpi_free_object to free handle and avoid above simulation issue. 

    diff -r 53d54ff605f2 sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp
    --- a/sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp      Tue Oct 22 18:18:11 2019 +0800
    +++ b/sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp      Wed Oct 23 01:13:53 2019 +0800
    @@ -98,9 +98,15 @@
                 LOG_CRITICAL("VPI: passed a NULL pointer : ABORTING");
             }
    
    +#ifdef VCS
    +        if (!(vpi_free_object(get_handle<vpiHandle>()))) {
    +            LOG_CRITICAL("VPI: unable to remove callback : ABORTING");
    +        }
    +#else
             if (!(vpi_remove_cb(get_handle<vpiHandle>()))) {
                 LOG_CRITICAL("VPI: unable to remove callback : ABORTING");
             }
    +#endif
    
             check_vpi_error();
         } else {

cmarqu, 2019-10-22:
Thanks for investigating the fix!
Do you have a short example that triggers the problem?
In your added code, I would also change the message slightly to "VPI: unable to free object".

speq, 2019-10-23:
Hi cmarqu,

I can't reproduce it on examples. It happens only on our whole chip simulation. However, once it happened, it continues crashing the simulator instead of showing Exception traceback in log. I guess it is a bug in vcs since it was mentioned on SolvNet.
Updating patch:

    diff -r db1bc9d87f68 sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp
    --- a/sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp      Tue Oct 22 17:50:04 2019 +0800
    +++ b/sim/cocotb/cocotb/share/lib/vpi/VpiCbHdl.cpp      Wed Oct 23 11:07:11 2019 +0800
    @@ -98,9 +98,15 @@
                 LOG_CRITICAL("VPI: passed a NULL pointer : ABORTING");
             }
    
    +#ifdef VCS
    +        if (!(vpi_free_object(get_handle<vpiHandle>()))) {
    +            LOG_CRITICAL("VPI: unable to free object : ABORTING");
    +        }
    +#else
             if (!(vpi_remove_cb(get_handle<vpiHandle>()))) {
                 LOG_CRITICAL("VPI: unable to remove callback : ABORTING");
             }
    +#endif
    
             check_vpi_error();
         } else {

jdanz37, 2019-10-23:
With VCS version, P-2019.06-1, I was seeing crashes with test_array, test_avalon, test_cocotb, and test_external. All were resolved with the above patch.

cmarqu, 2019-10-23:
Nice, I can confirm that this was present back until at least a 2016 version of VCS and this fixes it. @speq Can you please open a pull request for this patch?

eric-wieser, 2019-10-23:
I'm a little wary about adding an `ifdef VCS` into the vpi headers - it means that they need to be recompiled against each simulator, rather than reused across them. As far as I can tell, we don't have any such `ifdefs` today.

Can we inject this information at runtime instead of at compile-time?

Also, does it work if we call `vpi_release_handle` instead of `vpi_remove_cb`? The latter is deprecated.

eric-wieser, 2019-10-23:
Very relevant section of the PLI handbook:

https://books.google.co.uk/books?id=3pEMBwAAQBAJ&pg=PA205&lpg=PA205&dq=vpi_remove_cb+vs+vpi+release+handle&source=bl&ots=AYhg-UHFo8&sig=ACfU3U15MsDnuiU6cTcSIUjRW_BZX8BZ1Q&hl=en&sa=X&ved=2ahUKEwi5uYmyirPlAhUAQRUIHdJACEUQ6AEwAnoECAkQAQ#v=onepage&q=vpi_remove_cb%20vs%20vpi%20release%20handle&f=false

eric-wieser, 2019-10-23:
As I read it, the patch you suggest is incorrect - it means that `trigger.unprime()` will not actually unprime the trigger, and the callback will still be called.

Are we somehow upsetting VCS by removing the same callback twice?

cmarqu, 2019-10-23:
I can try running with ``COCOTB_SCHEDULER_DEBUG`` tomorrow unless someone beats me to it.

cmarqu, 2019-10-24:
I didn't get far, but I was surprised to see this message at the start:
```
Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.

     -.--ns ERROR    cocotb.gpi                                 VpiCbHdl.cpp:74   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbStartOfSimulation(11)
```

jdanz37, 2019-10-25:
I worked around that by adding cbk to the PLI tab file

```
diff --git a/cocotb/share/makefiles/simulators/Makefile.vcs b/cocotb/share/makefiles/simulators/Makefile.vcs
index 0444c50..e530dfa 100644
--- a/cocotb/share/makefiles/simulators/Makefile.vcs
+++ b/cocotb/share/makefiles/simulators/Makefile.vcs
@@ -61,7 +61,7 @@ endif
 # Can't do this using an argument, we have to create a PLI table file
 # enabling write access to the design
 $(SIM_BUILD)/pli.tab :
-       echo "acc+=rw,wn:*" > $@
+       echo "acc+=rw,wn,cbk:*" > $@

```

cmarqu, 2019-10-29:
For me, adding ``cbk`` didn't change anything (although the option looks needed), and also ``acc+2`` instead of ``acc+1`` further down had no effect.
But it looks at least that "The VPI function 'vpi_register_cb' is not supported in the compiler." _doesn't_ mean that ``vpi_register_cb`` is not supported at all, it's just not supported for ``cbStartOfSimulation`` (and ``cbEndOfSimulation``).

hashhsah, 2020-05-04:
I had the same issue, for *some* testcases. I later found that testcases not ending with a wait for clock edge suffers from this problem, with the exact error message quoted above. Possibly this is related to #1261 .

I was able to work around the problem, as described in https://github.com/cocotb/cocotb/issues/1261#issuecomment-623485150 .

rnaefen, 2020-07-16:
I confirm the proposed patch written by @speq worked for me with VCS Q-2020.03 with axi_lite_slave example.

but adding wait for clock edge has proposed by @cmarqu  has not effect on my side

anjhon, 2023-10-08:
> I worked around that by adding cbk to the PLI tab file
>
> ```
> diff --git a/cocotb/share/makefiles/simulators/Makefile.vcs b/cocotb/share/makefiles/simulators/Makefile.vcs
> index 0444c50..e530dfa 100644 [...]

this work for me！with VCS compile option plus "-debug_acess+all"。while I encoutered this  problem only  in pytest(using cocotb_test)， using makefile is OK.

cmarqu, 2023-10-11:
@anjhon It would be nice if you could open a pull request with this minimal change.

ktbarrett, 2025-10-17:
We are finally testing VCS in CI (version 2025.06) and do not see this issue. Try updating your version of VCS and cocotb and try again. If you are still having issues, open a new issue with a reproducible example.