cocotb/cocotb #1169: exception during restart in Riviera

s3bs, 2019-10-26:
Not sure if this is intended behavior or not, I get the following using the latest master as well as https://github.com/cocotb/cocotb/pull/1164

basically ran a test for 15us which is before the end of the test and then restart the simulation

```
restart
# COUT:  15000.00ns ERROR    cocotb.scheduler                            __init__.py:191  in _sim_event                      Failing test at simulator request before test run
 completion: Simulator shutdown prematurely
# COUT: Traceback (most recent call last):
# COUT:   File "/my_hidden_path/cocotb/cocotb/__init__.py", line 192, in _sim_event
# COUT:     scheduler.finish_scheduler(SimFailure(msg))
# COUT:   File "/my_hidden_path/cocotb/cocotb/scheduler.py", line 779, in finish_scheduler
# COUT:     self._test.abort(exc)
# COUT:   File "/my_hidden_path/cocotb/cocotb/decorators.py", line 285, in abort
# COUT:     return self._force_outcome(outcomes.Error(exc))
# COUT:   File "/my_hidden_path/cocotb/cocotb/decorators.py", line 271, in _force_outcome
# COUT:     cocotb.scheduler.unschedule(self)
# COUT:   File "my_hidden_path/cocotb/cocotb/scheduler.py", line 494, in unschedule
# COUT:     self.cleanup()
# COUT:   File "/my_hidden_path/cocotb/cocotb/scheduler.py", line 799, in cleanup
# COUT:     raise Exception("Cleanup() called outside of the main thread")
# COUT: Exception: Cleanup() called outside of the main thread
# COUT:  15000.00ns ERROR    cocotb.gpi                                  gpi_embed.c:405  in embed_sim_event                 Passing event to upper layer failed
# VSIM: Simulation has finished.
```

This is using Aldec Riviera-PRO version 2018.10.137.7135  on a RedHat 7 with python 2.7.7

Restart as such works though and i can rerun the simulation in the GUI

eric-wieser, 2019-11-19:
You restarted this manually in the gui?

jwprice100, 2023-02-16:
I believe I've been having the same issue since upgrading from Python 3.7.6. If the simulator ends the simulation, cocotb hangs. This can be replicated by running the simulation from the GUI for a shorter duration than the test. If one restarts the simulator OR ends the simulation from the TCL command interface then Riviera/cocotb hangs and must be forcibly closed. Is the root cause of this behavior understood?

closed 2024-02-07 by themperek