cocotb/cocotb #1170: cocotb-config segfaults when COCOTB_SIM env variable is set

tonywmnix, 2019-10-26:
I accidently left this env variable set in a terminal and my compilation broke. Traced it down to cocotb-config seg faulting with no indication. Narrowed it down to this env variable, so I tried to set it to 0 and same result. 
Perhaps a small change in cocotb/config.py to warn about this variable could help others.
Also, I would expect COCOTB_SIM=0 to have the effect of unsetting the variable entirely. 
thanks!

imphil, 2019-10-27:
Thanks for your bug report @tonywmnix! 

For clarification, that's the error message below:

```
$ COCOTB_SIM=1 cocotb-config 
Traceback (most recent call last):
  File "/home/philipp/.local/bin/cocotb-config", line 11, in <module>
    load_entry_point('cocotb', 'console_scripts', 'cocotb-config')()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/philipp/src/cocotb/cocotb/__init__.py", line 40, in <module>
    import cocotb.handle
  File "/home/philipp/src/cocotb/cocotb/handle.py", line 38, in <module>
    import simulator
ModuleNotFoundError: No module named 'simulator'
```

We could check in config.py for this environment variable and error out with a nicer error message.

tonywmnix, 2019-10-27:
The =1 is slightly misleading as it errors out for =0 as well.

themperek, 2020-04-16:
#1566

cmarqu, 2020-07-27:
Closing because this is resolved/removed as described in #1566.

closed 2020-07-27 by cmarqu