cocotb/cocotb #1175: Reconfigure the read-the-docs webhook

eric-wieser, 2019-10-30:
A week or two ago, I opted cocotb into the beta of doc PR builds

> On Thu, 17 Oct 2019 at 15:29, Santos from Read the Docs <santos.gallegos@read-the-docs-2cd3740a35e6.intercom-mail.com>
> Hi Eric, I have enabled the feature for the project. Let us know if you have any feedback of the feature!
> Santos from Read the Docs
>> On Thu, Oct 17, 2019 at 09:36 AM, "Eric Wieser" <wieser.eric@gmail.com> wrote:
>> Fantastic, thanks. Is there any chance you could do the same to cocotb.rtfd.io?
>>> On Wed, Oct 16, 2019 at 07:47 PM, "Eric Holscher" <eric.holscher@read-the-docs-2cd3740a35e6.intercom-mail.com> wrote: 
>>> Sure, I've just enabled it for you.
>>>
>>> Cheers,
>>> Eric
>>>
> > > > On Wed, Oct 16, 2019 at 05:12 PM, "Eric Wieser" <wieser.eric@gmail.com> wrote: 
> > > >  Hi!
> > > > 
> > > > I saw that building docs for PR's is an opt-in-by email feature, according to https://blog.readthedocs.com/building-docs-for-pull-requests/.
> > > > Would it be possible to opt in for ttps://clifford.readthedocs.io?
> > > > 
> > > > Thanks 
> > > > Eric Wieser

I found on my other project that to make this work, I had to completely remove the RTD webhook, and then re-add it through the ReadTheDocs website (via github auth). I don't have the power to do this, but I beleive @imphil does.

imphil, 2019-10-30:
I removed and re-added the webhook, let's see if that worked.

eric-wieser, 2019-10-30:
Thanks, working in #1172 :)

closed 2019-10-30 by eric-wieser