cocotb/cocotb #1176: Add docstring for fork()

cmarqu, 2019-10-30:
``fork`` is missing a docstring, and so it doesn't show up in https://cocotb.readthedocs.io/en/latest/library_reference.html.

cmarqu, 2019-10-30:
Okay, when I add ``.. autofunction:: cocotb.fork``, it does give me the docstring of ``scheduler.add`` (because ``fork`` is an alias for it) - but this should be rewritten to be useful for an end user. Right now, it's:

> Add a new coroutine.
> Just a wrapper around self.schedule which provides some debug and
> useful error messages in the event of common gotchas.

eric-wieser, 2019-11-04:
Leaving this open, since the docstring itself needs work

cmarqu, 2019-11-05:
> Leaving this open, since the docstring itself needs work

Indeed. It should also link to ``Join``.

ktbarrett, 2020-07-07:
Is this considered complete? Do we want just demos, or a more detailed docstring as well?

cmarqu, 2020-07-08:
Let's close this.
I think https://docs.cocotb.org/en/latest/coroutines.html should mention ``Join``, but that's for another issue.

closed 2020-07-08 by cmarqu