cocotb/cocotb #1181: Library naming and structuring discussion

cmarqu, 2019-11-03:
Pull Request #1168 renames ``libvhpi`` to ``libcocotbvhpi`` because Synopsys VCS-MX already ships a ``libvhpi``.
The PR spawned a discussion about library names in general; this issue collects the comments made there so that they don't get lost.

Technical points made so far were by @ktbarrett who wrote initially:
> Why not ``libcocotbvcs``? Since we have to compile the GPI using simulator-specific defines, ``libcocotbvhpi`` is misleading, as it might not work with another simulator with VHPI support. This is probably universal to the other objects as well.

and  later:

> Tool specific code is only an issue if it affects user code or user-facing code. Simulator-specific workarounds at the simulator interface level are unavoidable, but as long as the GPI abstracts them away, there is no problem. ``libvhpi.so`` and all the other simulator-to-GPI abstraction objects should be associativity named to the simulator they support. Then we do not have to rebuild an object every time we change simulators that use the same interface.

> Personally, I'd prefer ``libgpi.{SIM}.so`` seeing as these objects are a specific implementation of the GPI.

@themperek then wrote:
> When we are at this. Following #1062 I was proposing to compile everything at setup using setuptools (separate for every simulator). I can prepare this for cocotb it there is a will to move in this direction. It is implemented here: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/build_libs.py Can change naming conversion.

ktbarrett, 2019-11-04:
I want to add another point before further discussion. A particular simulator can have multiple interfaces. For example, Questa supports the FLI, VPI, and will soon support VHPI.

ktbarrett, 2019-11-04:
I like @themperek's idea of building all supported gpi implementations at setup and installing them to the venv. We could add a script to the distribution that would return information about the build and installation for those that don't use the makefiles for their projects.

eric-wieser, 2019-11-20:
> A particular simulator can have multiple interfaces.

Perhaps then `libgpi_{SIM}_{which_pi}.so`?

I don't see a lot of precedence for using `.` within library names, so would be inclined not to

ktbarrett, 2020-01-07:
I have been thinking for a while about this, since before this discussion. In fact I have somewhere a fork of cocotb where I replaced all the building makefiles with CMake and introduced a python script for running simulators with cocotb rather than using makefiles for run. You would simply run CMake specifying the simulators you wanted to support (`cmake .. -SIMS="questa,icarus,ghdl"`), building (`make -j`) would generate `libvpi.questa.so`, `libfli.questa.so`, etc. and you would install them to the system (or wherever via `CMAKE_INSTALL_PREFIX` and `make install`). The python script inplemented many cocotb variables (like `TESTCASE`) and passed any other arguments to the simulator call: `cocotb --sim questa --module modulename --testcase testcase +any -other="args"`.

If this is a good idea, it definitely accomplishable.

I think we should build and install the GPI and all the implementations ahead of time to the system. One advantage is that you get the ability to run multiple simulators without having to rebuild your libraries. I have said before that the GPI *could* be useful in contexts outside of cocotb, *if* effort was put into refactoring the interface to make it an independent entity. I can imagine that someone might want to wrap the GPI in another language interface. There was a user on the gitter this week asking for a Java port of cocotb; and personally, I'd like to see a Go port.

themperek, 2020-01-07:
@ktbarrett Another idea (that seems works) was to use standard python packaging you can look here: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/build_libs.py (this need some naming/code adjustment but works). Libraries are compiled on setup. We can distribute binary on pipy. I was planning to refactor this a bit and propose PR.

ktbarrett, 2020-01-07:
@themperek I'm not opposed to any build tool, Python makes sense. I think the build system and run script you have would be a great way to replace the makefile flow. I would do a PR.

At some though, I'd like to redraw the library boundaries so they are less coupled. `libgpi` shouldn't link against `libcocotb`, `libcocotbutils` can be header only, etc. I'd like to make GPI a consumable standalone library.

mballance, 2020-01-08:
I would hope that, in the case of a standard interface (eg VPI), a single library could be built to support all compliant simulators. Perhaps a 'standard' interface-specific library (eg libcocotb.vpi.so) could be provided and, if there is an outlier environment, then a specific library could be provided for that environment (eg libcocotb.strange.vpi.so for the 'strange' tool).
Especially when it comes to VPI, I'd be happy to volunteer my time to analyze and work on fixing any current simulator-specific aspects that the group is aware of.

@ktbarrett, funny enough, I also have a fork of cocotb whose goal is to simplify running simulators with cocotb instead of running the simulators under the cocotb makefiles. In addition to adding a standalone Makefile to build the libraries, I added the option of controlling cocotb via plusargs (eg enabling control of the testcase via +cocotb.testcase=MY_TEST in addition to using the TESTCASE environment variable).

mballance, 2020-01-11:
One thing I've been wondering about, which is related to this library-naming question, is how much of the code that is currently C/C++ could actually be written in Python. With ctypes, it seems that much of the interface code (gpi and the simulator-specific code) could be written in Python. The library that the simulator initially loads would need to be written in C/C++, but that (I think) could be an incredibly small amount of code.
Does this seem interesting to pursue?

imphil, 2020-02-16:
I've filed https://github.com/cocotb/cocotb/issues/1392 to track the progress towards a single simulator interface library.

closed 2020-04-04 by eric-wieser