cocotb/cocotb #1182: The sysconfig.get_config_var("LDLIBRARY") returns wrong value, a static Python library

tymonx, 2019-11-04:
Issue:
```
# Loading sv_std.std
# Loading work.endian_swapper_sv(fast)
# Loading cocotb/examples/endian_swapper/tests/build/libs/x86_64/libvpi.so
# Unable to open lib libpython3.7m.a (libpython3.7m.a: invalid ELF header)
#      -.--ns ERROR    cocotb.gpi                                  gpi_embed.c:139  in embed_init_python               Failed to find Python shared library
#
#      -.--ns INFO     cocotb.gpi                                  gpi_embed.c:102  in set_program_name_in_venv        Using Python virtual environment interpreter at cocotb_test/env/bin/python
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
#      -.--ns ERROR    cocotb.gpi                                  gpi_embed.c:196  in get_module_ref                  Failed to load Python module "cocotb"
#
# End time: 10:09:08 on Nov 04,2019, Elapsed time: 0:00:07
# Errors: 0, Warnings: 0
```

Root cause in the `cocotb/share/makefiles/Makefile.pylib.Linux` file:
```
PYTHON_LIBDIR:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("LIBDIR"))')
```

We have both `libpython3.7m.a` and `libpython3.7m.so` libraries in the same location but the `sysconfig.get_config_var("LDLIBRARY")` call returns the `libpython3.7m.a`

Quick workaround in the `cocotb/share/makefiles/Makefile.pylib.Linux` file that doesn't effect the `.so` extension:
```
PYTHON_LIBDIR:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("LIBDIR").replace(".a", ".so"))')
```

tymonx, 2019-11-04:
It seems that this behavior is expected when the Python was configured and built with a disabled shared library support during the `./configure --disable-shared` step. In that case the `LDLIBRARY` variable can point to a Python static library and still you can have a Python shared library under a Python library path. Relying only on the `LDLIBRARY` variable value is not good enough.

https://github.com/python/cpython/blob/master/configure#L5870

```
{ $as_echo "$as_me:${as_lineno-$LINENO}: checking LIBRARY" >&5
$as_echo_n "checking LIBRARY... " >&6; }
if test -z "$LIBRARY"
then
	LIBRARY='libpython$(VERSION)$(ABIFLAGS).a'
fi
{ $as_echo "$as_me:${as_lineno-$LINENO}: result: $LIBRARY" >&5
$as_echo "$LIBRARY" >&6; }

LDLIBRARY="$LIBRARY"

# Other platforms follow
if test $enable_shared = "yes"; then
  PY_ENABLE_SHARED=1

$as_echo "#define Py_ENABLE_SHARED 1" >>confdefs.h

  case $ac_sys_system in
    Linux*|GNU*|NetBSD*|FreeBSD*|DragonFly*|OpenBSD*)
	  LDLIBRARY='libpython$(LDVERSION).so'
	  BLDLIBRARY='-L. -lpython$(LDVERSION)'
	  RUNSHARED=LD_LIBRARY_PATH=`pwd`${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
	  INSTSONAME="$LDLIBRARY".$SOVERSION
	  if test "$with_pydebug" != yes
          then
	      PY3LIBRARY=libpython3.so
	  fi
	  ;;
  esac
else # shared is disabled
  PY_ENABLE_SHARED=0
  case $ac_sys_system in
    CYGWIN*)
          BLDLIBRARY='$(LIBRARY)'
          LDLIBRARY='libpython$(LDVERSION).dll.a'
          ;;
  esac
fi
```

imphil, 2019-11-04:
You need to use a python version compiled with `--enable-shared`; if that's not the case, you need to recompile. Make sure that you don't install both variants into the same directory, that will lead to a confusingly broken python installation.

tymonx, 2019-11-04:
Now I known that but I cannot do that. We are working on fixed compute engineering machines without a root access and it cannot be managed by us. Python is already installed on these machines with the `--disable-shared` option. But both Python libraries are available under the same library path. I think that the **cocotb** shouldn't relay on the **LDLIBRARY** variable because it simply fails in that case. And the workaround (replacing `.a` with `.so` extension) proofs that it can work in both cases.

tymonx, 2019-11-04:
@imphil I can make a Pull Request with that:
```
PYTHON_LIBDIR:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("LIBDIR").replace(".a", ".so"))')
```

It shouldn't affect current cocotb behavior for Python configured with the `--enable-shared` option. But I want to know your opinion about that.

closed 2019-11-19 by imphil

tsengr0916, 2019-12-10:
could COCOTB support the static Python library (libpython*.a) instead of the dynamic version? Our IT department does not know how to install the .so file. They say it does not come with for Python 3.6.

imphil, 2019-12-11:
No it cannot. You need to use a Python version compiled with `--enable-shared`. Actually, the IT department should not compile python itself, but use distribution packages.

tsengr0916, 2019-12-11:
Thanks. I'm not sure if I fully understand this. What do you really mean a python version compiled with --enable-shared? How do I tell if a Python version has been compiled with the switch? Are you talking about the Python needs to be compiled with the C code that interfaced with a simulator? 
Don't all the Python versions come with the .so file?

tsengr0916, 2020-01-03:
How do I tell which Python version is compiled with -enable-shared? Does the latest version 3.8.1 supported by COCOTB?