cocotb/cocotb #1189: Does COCOTB work for Python 3.6.4?

tsengr0916, 2019-11-15:
Hi,

I downloaded the latest cocotb and installed with Python 3.6.4, but it does not work:

gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -Wstrict-prototypes -Waggregate-return -DFILTER -DMODELSIM -shared -Xlinker -L/.pkg/qct/software/python/3.6.4/lib -fPIC -o /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/pandace/inc/libs/blocks/pandace/verif/tmp/cocotb/examples/endian_swapper/tests/build/libs/x86_64/libgpilog.so /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/pandace/inc/libs/blocks/pandace/verif/tmp/cocotb/examples/endian_swapper/tests/build/obj/x86_64/gpi_logging.o   -lpython3.6m 
/usr/lib64/gcc/x86_64-suse-linux/4.3/../../../../x86_64-suse-linux/bin/ld: /.pkg/qct/software/python/3.6.4/lib/libpython3.6m.a(abstract.o): relocation R_X86_64_32S against `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/.pkg/qct/software/python/3.6.4/lib/libpython3.6m.a: error adding symbols: Bad value
collect2: ld returned 1 exit status

If I switch to Python 3.5.2, then it works fine. Any ideas? 

Thanks,
Richard

cmarqu, 2019-11-16:
In case you have both a static and a library installed for Python 3.6.4, does #1182 help?

tsengr0916, 2020-01-03:
Hi imphil,
In the #1182, you mentioned that "You need to use a Python version compiled with --enable-shared. Actually, the IT department should not compile python itself, but use distribution packages."

But how do I tell which Python version is compiled with -enable-shared? Does the latest version 3.8.1 supported by COCOTB?

Thanks!
Richard

imphil, 2020-01-03:
Yes, python 3.8.1 is supported by cocotb. 

To check how Python was compiled refer to this answer: https://stackoverflow.com/questions/23201498/how-to-check-if-python3-was-built-with-enable-shared

tsengr0916, 2020-01-04:
Many thanks!!
Richard

closed 2020-01-04 by imphil