cocotb/cocotb #1194: warning with DeprecationWarning does not give output.

Fatsie, 2019-11-19:
Using ``DeprecationWarning`` seems to suppress the warning.
I have the following test:
```python
import cocotb
import warnings
from cocotb.triggers import Timer

@cocotb.test()
def deprecation_warning(dut):
    warnings.warn(
        "With DeprecationWarning",
        DeprecationWarning, stacklevel=2
    )
    warnings.warn(
        "Without DeprecationWarning",
        stacklevel=2
    )
    yield Timer(1)
```
Given the following output:
```console
[verhaegs@localhost test_deprecation]$ make
make results.xml
make[1]: Map '/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation' wordt binnengegaan
PYTHONPATH=/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation/build/libs/x86_64:/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation:/home/verhaegs/eda/code/cocotb: LD_LIBRARY_PATH=/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation/build/libs/x86_64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64::/usr/lib64:/usr/lib64:/usr/lib64:/usr/lib64 MODULE=test_deprecation \
        TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /home/verhaegs/software/bin/vvp -M /home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:303  in embed_sim_init                  Running on Icarus Verilog version 11.0 (devel)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:304  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:133  in _initialise_testbench           Running tests with Cocotb v1.2.0 from /home/verhaegs/eda/code/cocotb
     0.00ns INFO     cocotb                                      __init__.py:150  in _initialise_testbench           Seeding Python random module with 1574178915
     0.00ns INFO     cocotb.regression                         regression.py:190  in initialise                      Found test test_deprecation.deprecation_warning
     0.00ns INFO     cocotb.regression                         regression.py:343  in execute                         Running test 1/1: deprecation_warning
     0.00ns INFO     ..eprecation_warning.0x7f798905f210       decorators.py:251  in _advance                        Starting test: "deprecation_warning"
                                                                                                                     Description: None
/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation/test_deprecation.py:9: UserWarning: Without DeprecationWarning
  warnings.warn("Without DeprecationWarning")
VCD info: dumpfile waveform.vcd opened for output.
     0.00ns INFO     cocotb.regression                         regression.py:288  in handle_result                   Test Passed: deprecation_warning
     0.00ns INFO     cocotb.regression                         regression.py:213  in tear_down                       Passed 1 tests (0 skipped)
     0.00ns INFO     cocotb.regression                         regression.py:395  in _log_test_summary               **********************************************************************************************
                                                                                                                     ** TEST                                  PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     **********************************************************************************************
                                                                                                                     ** test_deprecation.deprecation_warning    PASS            0.00          0.00         2.16  **
                                                                                                                     **********************************************************************************************
                                                                                                                     
     0.00ns INFO     cocotb.regression                         regression.py:412  in _log_sim_summary                *************************************************************************************
                                                                                                                     **                                 ERRORS : 0                                      **
                                                                                                                     *************************************************************************************
                                                                                                                     **                               SIM TIME : 0.00 NS                                **
                                                                                                                     **                              REAL TIME : 0.00 S                                 **
                                                                                                                     **                        SIM / REAL TIME : 0.72 NS/S                              **
                                                                                                                     *************************************************************************************
                                                                                                                     
     0.00ns INFO     cocotb.regression                         regression.py:222  in tear_down                       Shutting down...
make[1]: Map '/home/verhaegs/eda/code/cocotb/tests/test_cases/test_deprecation' wordt verlaten
```
The warning with DeprecationWarning is not seen.

eric-wieser, 2019-11-19:
This is just the default python behavior. Related is https://www.python.org/dev/peps/pep-0565/.

eric-wieser, 2019-11-19:
Based on that, it sounds like we should add to the regression manager,
```
warnings.filterwarnings("default", category=DeprecationWarning,
                                   module=the_test_module_name)
```

closed 2019-12-05 by eric-wieser