cocotb/cocotb #1195: Document supported C and C++ version

imphil, 2019-11-19:
From https://github.com/cocotb/cocotb/pull/1149#issuecomment-545019530

> If you are going to call out the C++ standard, it's a good idea to call out the C standard as well. GCC 4.8.1 does not fully support C11, so the C standard should be gnu99. The GNU extensions are needed to enable POSIX standards required by Python, and are supported by Clang as well.

Documentation should be added to https://cocotb.readthedocs.io/en/latest/quickstart.html

We currently specify the minimum supported GCC version (intentionally), since that's the compiler we need to be compatible (see #1087 for details).

While I'd like to keep specifying the minimum compiler version (since that's what matters for compat), we should add the C and C++ versions to make it obvious what can be used in software:

- For C++, it's C++11 ("GCC 4.8.1 was the first feature-complete implementation of the 2011 C++ standard, previously known as C++0x.", https://gcc.gnu.org/projects/cxx-status.html)
- For C, it's gnu99 (1)

(1) Full C11 might never happen if we want to stay compatible with MSVC, but that's a very different story for another day.

ktbarrett, 2019-11-21:
#1137 adds `-std=gnu99` to the makefiles and mentions that [Clang 3.3+](https://clang.llvm.org/cxx_status.html) is supported.

eric-wieser, 2020-06-18:
@imphil, probably worth doing this for the release - I don't remember whether the requirements have changed since 1.3.

ktbarrett, 2020-06-18:
This is probably only necessary in the case of C++. On Linux LD_LIBRARY_PATH has to point to a version of libstdc++ that support C++11. I've run into this problem before when using really old version of Questa, because it prepends it's libraries to the LD_LIBRARY_PATH. Recently we had #1933 who was running an old OS with a screwed up environment. And that will be an issue once (if?) we start distributing binary packages and people try running cocotb on old OSes (like RHEL6). It would no longer fail to install, but it wouldn't run :)

PR #1936 (2020-06-22): C++11 docs
Closes #1195. Specifically calls out C++11 compiler and adds documentation to the troubleshooting section on debugging issues related to loading the wrong libstdc++ (which is somewhat common).
closed 2020-06-22 by imphil