cocotb/cocotb #1197: Time to revise commit best practice guideline ?

Fatsie, 2019-11-20:
In #1190 I was asked to comply to the commit message best practice of 50 char summary and body with max. line length of 72.
Skimming through latest merges on master, only a minority of the commit messages seem to follow this guideline. I have even seen markdown type commit messages with one long line for each paragraph.
Maybe time to update contribution guideline to evolving commit message best practice ?

imphil, 2019-11-20:
Best practice hasn't changed, but I'm slightly flexible in enforcement (and sometimes miss things).

tymonx, 2019-11-20:
You can add the [commitlint](https://github.com/conventional-changelog/commitlint) as a pre gate keeper for git commits in current CI flow.

eric-wieser, 2019-11-20:
> I have even seen markdown type commit messages with one long line for each paragraph.

I'm often guilty of this - markdown is convenient for commit messages, as it pastes directly into the github ui, but is still very readable in plaintext; and I find hard-wrapping prose frustrating, as it makes editing the middle of a paragraph entail rewrapping - something that I'd rather leave to the viewer.

It's not my call to make though, just explaining why when given the choice I do it this way

Fatsie, 2019-11-20:
IMHO this guideline is a best practice of times before there was github or even gitk.
I do think this comes from the times people were working on terminal with 80 characters line width. I doubt there are much cocotb developers/users with such a setup. Looking at other project I am following I also don't think this rule is considered best practice anymore.
Personally I prefer a litlte longer single line summary over a summary that does not tell much with a single line body.

closed 2025-10-16 by ktbarrett