cocotb/cocotb #1202: IntegerObject: Document and ensure its 32 bit

imphil, 2019-11-20:
`IntegerObject` is used to set int/integer nets as defined in SystemVerilog, which are 32 bit (signed) values. See https://github.com/cocotb/cocotb/issues/268 for a discussion on signed-ness.

- `IntegerObject` doesn't check the size, which could lead to loss of data.
- We need to document that `IntegerObject` is 32 bit.
- Our backend functionality, starting with `simulator.set_signal_val_long`, uses long instead of int as data type and in the function names. That should be cleaned up to say int.

elgorwi, 2020-04-30:
Looking into this issue and just opened an initial PR. I developed this using Modelsim (32-bit) on windows Msys2 platform. Initial compilation of the C++ libs (no local changes) revealed an issue, that is probably caused by using a 32-bit platform. I compiled this on OSX (64-bit platform) as well without any compilation warnings:
```
    cocotb/share/lib/vhpi/VhpiCbHdl.cpp: In member function 'virtual int VhpiSignalObjHdl::set_signal_value(long int, gpi_set_action_t)':
    cocotb/share/lib/vhpi/VhpiCbHdl.cpp:579:24: warning: comparison of integer expressions of different signedness: 'long int' and 'unsigned int' [-Wsign-compare]
      579 |             if ((value > EnumLimits::max()) || (value < EnumLimits::min())) {
          |                  ~~~~~~^~~~~~~~~~~~~~~~~~~
    cocotb/share/lib/vhpi/VhpiCbHdl.cpp:579:55: warning: comparison of integer expressions of different signedness: 'long int' and 'unsigned int' [-Wsign-compare]
      579 |             if ((value > EnumLimits::max()) || (value < EnumLimits::min())) {
          |                                                 ~~~~~~^~~~~~~~~~~~~~~~~~~
    cocotb/share/lib/vhpi/VhpiCbHdl.cpp:589:24: warning: comparison of integer expressions of different signedness: 'long int' and 'unsigned int' [-Wsign-compare]
      589 |             if ((value > IntLimits::max()) || (value < IntLimits::min())) {
          |                  ~~~~~~^~~~~~~~~~~~~~~~~~
    cocotb/share/lib/vhpi/VhpiCbHdl.cpp:589:54: warning: comparison of integer expressions of different signedness: 'long int' and 'unsigned int' [-Wsign-compare]
      589 |             if ((value > IntLimits::max()) || (value < IntLimits::min())) {
          |                                                ~~~~~~^~~~~~~~~~~~~~~~~~
```
This warning is relevant when changing ```simulator.set_signal_val_long``` to only accept 32-bit values. It looks like "long" is not the same size on 32-bit and 64-bit platforms. 
Interestingly, vhpi is expecting an unsigned int```typedef uint32_t vhpiIntT;``` where vpi is expecting a signed int ```typedef int             PLI_INT32;```
Should we cast the value checks VhpiCbHdl.cpp to int64_t to get rid of the warnings above? Or should we remove the value checks altogether because we would only be accepting signed 32-bit values through ```simulator.set_signal_val_long``` anyway?  
I don't have access to a vhpi simulator to see if a direct cast would cause any undesired side effects.

ktbarrett, 2020-04-30:
@elgorwi You may also look at #268. There is some discussion on that.

`long` means different things on different systems, on 32 bit systems and 64 bit Windows `long` is 32 bit, most everywhere else they are 64 bit. To err on the safe side we really should be using `int32_t` for `set_signal_value_long`. In the case of assigning to Verilog `integer`s or VHDL `integer`/`natural` etc. we are limited to 32 bit integers anyway. Assuming that `set_signal_value_long` is how we appropriately set integer signals, supporting more than 32 bits is likely bad.

One potential issue is how to support integer types that *are* 64 bit integers (I think Verilog 2005 changed integers to 64 bit, but not many simulators followed suit).

Were you planning on making this change in the already open PR?

elgorwi, 2020-05-01:
> @elgorwi You may also look at #268. There is some discussion on that.
>
> `long` means different things on different systems, on 32 bit systems and 64 bit Windows `long` is 32 bit, most everywhere else [...]

@ktbarrett Thanks for the input. I'll open another pull request to handle changes to ```simulator.set_signal_val_long```

elgorwi, 2021-01-09:
Fixed by #2316

PR #2316 (2021-01-09): set_signal_value_long() bugfixes and improvements
Some bugfixes and improvements for setting signals using python ints: 
closes #1202 
closes #913 
closes #268 

This PR is essentially a merge of PR #1792 , PR #1742  and #1744 .
For more details refer to the linked issues and PRs.
closed 2021-01-09 by ktbarrett