cocotb/cocotb #1210: Cannot drive weak states on DUT signals

sfavazza-duagon, 2019-11-25:
Hello,

I am simulating a bus similar to an I2C bus, where more than one driver can act on a bus line.
In my test bench I defined a driver which send data by forcing one of the bus lines to state '1' or '0'.
When the data transfer is over I must leave the lines in weak pull-down state to allow the DUT to respond.

I tried the following:
```
bus_line <= BinaryValue('w')
```
But the line remains to the last assigned value (either strong one '1' or strong zero '0').
I also tried to drive the line to weak high `h` and weak `low`, then the line is forced to the strong versions respectively.

Digging in the `BinaryValue` definitions I've seen that `l` and `h` are unconditionally resolved to the strong state versions.
Is it the intended behavior?
Is there another approach to drive weak states for the DUT signals?

Perhaps I might work on the issue if this case is not addressed in the current version of `cocotb`.

eric-wieser, 2019-11-25:
When I faced this problem, I ended up adding a weak driver to my test-bench, and then controlling that via cocotb.

sfavazza-duagon, 2019-11-25:
Can you post a quick example?

eric-wieser, 2019-11-25:
Only from memory, it was something like:
````verilog
module my_tb;

wire sda_en;  // drive with cocotb
wire sda = sda_en ? 1'b0 : 1'bz;

my_dut dut(.sda(sda));

endmodule
``

sfavazza-duagon, 2019-11-25:
I see,

thank you for the fast answer.

By the way, would it be easier to make `BinaryValue` supporting also the weak values?
I would like to understand whether there is a practical reason why it was decided to resolve them to strong values.

ktbarrett, 2019-11-25:
Unfortunately, that is not how cocotb works. cocotb can only force or deposit a value onto a bus discretely, it cannot dynamically add another driver. Eric's solution is the only solution when dealing with tri-state stuff. I don't know if this limitation can be surpassed.

Related to #1083.

eric-wieser, 2019-11-25:
Is that a vpi limitation?

closed 2019-11-26 by sfavazza-duagon

cmarqu, 2019-12-02:
``s_vpi_value.format`` can also be ``vpiStrengthVal``, so maybe we "just" need to add support for this around https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/vpi/VpiCbHdl.cpp#L325? The PLI Handbook describes this on pg. 155ff.

ktbarrett, 2019-12-03:
@cmarqu What does it say? I saw that, but I don't have the book to investigate. It's kind of expensive...

cmarqu, 2019-12-03:
I also don't have the book, but a bit of strategic websearching digs up links with some sample pages.

ktbarrett, 2019-12-03:
I have a feeling the answer is on page 156, which is conveniently greyed-out.

@eric-wieser Correct me if I am mistaken, but you have a copy of The Verilog PLI Handbook?

eric-wieser, 2019-12-03:
I'd hoped I'd have institutional access, but seem not to. I have heard that dubiously legal PDFs exist online on russian websites though.

eric-wieser, 2019-12-03:
Unfortunately, the book is pre verilog 2009, which means it's missing some of the vpi triggers that I think cocotb should be using.