cocotb/cocotb #1211: ParametrizedSingleton not so effective

Fatsie, 2019-11-26:
I modified code for ParametrizedSingleton  a little bit to give a message when a KeyError happens:
```python
class ParametrizedSingleton(type)
...
    def __call__(cls, *args, **kwargs):
        key = cls.__singleton_key__(*args, **kwargs)
        try:
            return cls.__instances[key]
        except KeyError:
            # construct the object as normal
            print("KeyError in {!r}: key={!r}".format(cls, key))
            self = super(ParametrizedSingleton, cls).__call__(*args, **kwargs)
            cls.__instances[key] = self
            return self
```
I made following test bench:
```python
import cocotb
from cocotb.triggers import ReadOnly, RisingEdge
from cocotb.clock import Clock

@cocotb.coroutine
def loop1(clk, n):
    for _ in range(n):
        yield RisingEdge(clk)

@cocotb.coroutine
def loop2(clk, n):
    for _ in range(n):
        yield ReadOnly()
        yield RisingEdge(clk)

@cocotb.test()
def test_bed3(dut):

    cocotb.fork(Clock(dut.clk, 10, 'ns').start())

    print("loop1")

    yield loop1(dut.clk, 10)

    print("loop2")

    yield loop2(dut.clk, 10)
```
It gives the following output:
```console
make results.xml
make[1]: Map '/home/verhaegs/eda/code/cocotb/tests/test_cases/test_bed' wordt binnengegaan
PYTHONPATH=/home/verhaegs/eda/code/cocotb/tests/test_cases/test_bed/build/libs/x86_64:/home/verhaegs/eda/code/cocotb/tests/test_cases/test_bed:/home/verhaegs/eda/code/cocotb: LD_LIBRARY_PATH=/home/verhaegs/eda/code/cocotb/tests/test_cases/test_bed/build/libs/x86_64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64::/usr/lib64:/usr/lib64:/usr/lib64:/usr/lib64 MODULE=test_bed3 \
        TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /home/verhaegs/software/bin/vvp -M /home/verhaegs/eda/code/cocotb/tests/test_cases/test_bed/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
KeyError in <class 'cocotb.triggers.NextTimeStep'>: key=None
KeyError in <class 'cocotb.triggers.ReadWrite'>: key=None
KeyError in <class 'cocotb.triggers.ReadOnly'>: key=None
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:347  in embed_sim_init                  Running on Icarus Verilog version 11.0 (devel)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:348  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with cocotb v1.2.0 from /home/verhaegs/eda/code/cocotb
     0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1574786511
     0.00ns INFO     cocotb.regression                         regression.py:190  in initialise                      Found test test_bed3.test_bed3
     0.00ns INFO     cocotb.regression                         regression.py:348  in execute                         Running test 1/1: test_bed3
     0.00ns INFO     ..otb.test.test_bed3.0x7f56c4cec550       decorators.py:253  in _advance                        Starting test: "test_bed3"
                                                                                                                     Description: None
loop1
KeyError in <class 'cocotb.triggers.Join'>: key=<cocotb.decorators.RunningCoroutine object at 0x7f56c4ceca10>
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
VCD info: dumpfile waveform.vcd opened for output.
loop2
KeyError in <class 'cocotb.triggers.Join'>: key=<cocotb.decorators.RunningCoroutine object at 0x7f56c4cecb50>
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.RisingEdge'>: key=ModifiableObject(sample_module.clk)
KeyError in <class 'cocotb.triggers.Join'>: key=<cocotb.decorators.RunningCoroutine object at 0x7f56c4cecad0>
KeyError in <class 'cocotb.triggers.Join'>: key=<cocotb.decorators.RunningCoroutine object at 0x7f56c4cec9d0>
   190.00ns INFO     cocotb.regression                         regression.py:288  in handle_result                   Test Passed: test_bed3
   190.00ns INFO     cocotb.regression                         regression.py:213  in tear_down                       Passed 1 tests (0 skipped)
   190.00ns INFO     cocotb.regression                         regression.py:400  in _log_test_summary               *****************************************************************************
                                                                                                                     ** TEST                 PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     *****************************************************************************
                                                                                                                     ** test_bed3.test_bed3    PASS          190.00          0.01     14406.98  **
                                                                                                                     *****************************************************************************
                                                                                                                     
   190.00ns INFO     cocotb.regression                         regression.py:417  in _log_sim_summary                *************************************************************************************
                                                                                                                     **                                 ERRORS : 0                                      **
                                                                                                                     *************************************************************************************
                                                                                                                     **                               SIM TIME : 190.00 NS                              **
                                                                                                                     **                              REAL TIME : 0.02 S                                 **
                                                                                                                     **                        SIM / REAL TIME : 12354.61 NS/S                          **
                                                                                                                     *************************************************************************************
                                                                                                                     
   190.00ns INFO     cocotb.regression                         regression.py:222  in tear_down                       Shutting down...
```
So for ``loop1`` where only ClockEdge is used all is honky dory but for ``loop2`` where a ``ReadOnly()`` is added the ``RisindEdge(clk)`` is garbage collected at each loop iteration.
This due to the use of weak reference for caching. I see same behavior for python 2.7 and python 3.7.

eric-wieser, 2019-11-26:
So, the main intent of the caching was to guarantee that it wasn't possible to create two of the same instance, not so much for performance reasons.

eric-wieser, 2019-11-26:
For `Join`, the weakness was required to solve a reference cycle. In principle we could use a strong reference for the edges, since signals are never destroyed anyway

Fatsie, 2019-11-26:
So the question is, should it also be used for performance optimization or should user code still cache the event in a local variable before a loop ?

eric-wieser, 2019-11-26:
As long as we can optimize performance without introducing any non-benign reference cycles, then it should be ok.

Note that in the past, the edge triggers were actually cached on the signals as `_r_edge`, `_f_edge` etc. The removal of this was really more a side effect of fixing `type(RisingEdge(sig)) is RisingEdge` to be `true`. In principle, we could go back to storing a hard cache for:
* Simulator phase triggers
* Signal triggers

closed 2021-03-14 by github-actions[bot]