cocotb/cocotb #1212: Why is the native logger overridden?

ktbarrett, 2019-11-26:
https://github.com/cocotb/cocotb/blob/ba0fb097efbd2939ad6a961ee8d02793104ef8f7/cocotb/__init__.py#L60

How about instead of forcing users to use the cocotb logger through `logging`, we just extend `cocotb.log` to act more like the logging module and get users to use that instead?

eric-wieser, 2019-11-26:
This is already a lot simpler than it used to be.

If you look at the class, it doesn't really do very much. The most important thing is the _logFromC method.

I suppose that could be a free function, there's not a particularly strong argument for it to be a method.

ktbarrett, 2019-11-26:
That's not really what I'm asking. I'm asking why that quoted line is there. Is there any reason to set the logger class in `logging`? I see in a lot of files there are lines like

```python
import logging
testlog = logging.getLogger('cocotb.test')
```

Why override `logging` when you could instead just set up `cocotb.log` to act just like the `logging` module, so that you could do

```python
import cocotb.log
cocotb.log.getLogger('test')
```

Why replace the world, instead of coexist? Sometimes people want to use the `logging` module for things that aren't a simulation status/debugging logfile.

eric-wieser, 2019-11-26:
> Is there any reason to set the logger class in logging?

Here's why we do it today:

* To set a default logging formatter / handler
* To provide a hook to allow our C code to call into the python logging stuff
* To add some random color attributes

The first two of these would be possible without setting the logging class. The third isn't, but arguably isn't very useful.

stuarthodgson, 2019-11-26:
The last point is quite useful for me at least.

eric-wieser, 2019-11-26:
@stuarthodgson: How would you feel about deprecating `log.colour` in favor of just calling `cocotb.utils.want_color_output()` directly?

eric-wieser, 2019-11-26:
@ktbarrett, back to your original question:

> How about instead of forcing users to use the cocotb logger

Are you complaining that users can't now specify their own handler class?  There's nothing to stop a user doing:
```
@logging.setLoggerClass
class MyLog(logging.getLoggerClass()):
   pass  # this will work both with and without cocotb
```

Or are you complaining about the fact we set a default handler / formatter for all loggers?

ktbarrett, 2019-12-03:
> Or are you complaining about the fact we set a default handler / formatter for all loggers?

Yes. To be less round-about, we have our own logger that we use. We want to keep the format consistent with the rest of our software. We used to maintain a fork of cocotb that had the changes in `log.py` we wanted, and they were rather extensive. With the not-so-recent changes to the logger, which were also rather extensive, merging is not possible. We will have to start from scratch, so ignore my complaints.

ktbarrett, 2019-12-03:
Likely we will provide our own logger class and `SimLog` implementation and just change the import on this line.
https://github.com/cocotb/cocotb/blob/17ce4e51da43b6d2d1b5834d629374ed5535a8d8/cocotb/__init__.py#L42

eric-wieser, 2019-12-03:
I think I'd like to keep this open - the points about the class being unnecessary are still valid, and we could look at making the default config disableable by an env bar.

You might find that you don't need to fork cocotb at all, but instead can just revert it's setting changes at startup.

If you're able to share it, an outline of your old and/or new patch would be insightful to working out the best way for cocotb to stay out of your way.

ktbarrett, 2019-12-04:
I think the best way forward is to move logger setup out of `__init__.py` and into the `cocotb.log` module. `__init__.py` would first try to load a user's custom logger module (by an established name) and if `ModuleNotFoundError`, it would load `cocotb.log` as a fallback. A custom logger will obviously have to play nicely with how cocotb currently uses the logger. Maybe some documentation on required interfaces.

I'd like to see this pattern be applied to the embed_gpi/cocotb module entry point as well, so we could overload the entire (or at least part) of the setup to support launching into IPython.

eric-wieser, 2019-12-04:
> would first try to load a user's custom logger module

I think we just leave this to the python logging infrastructure.

> I'd like to see this pattern be applied to the embed_gpi/cocotb module entry point as well, so we could overload the entire (or at least part) of the setup to support launching into IPython.

Filed #1225 for that to keep discussion here focused

> Maybe some documentation on required interfaces.

Once #1213 is in, the answer will basically be "there is no cocotb logging interface, just configure the python logging module as you see fit".

ktbarrett, 2019-12-04:
We need the ability to inject a Logger class before the cocotb, gpi, scheduler, and regression loggers are constructed, or there will be inconsistent logging behavior, no? Currently no user code is run until the user's test module is loaded.

eric-wieser, 2019-12-27:
> We need the ability to inject a Logger class

I'd dispute the claim that anyone really needs a custom logger class - most use cases are solved with a custom handler. Unfortunately our custom class is so hostile today that we force users to subclass it to stop our handlers being installed. That's fixed in gh-1266.

eric-wieser, 2020-02-03:
@ktbarrett, closing this because the answer to

> Why is the default logger overridden?

Is now just

> To provide attributes that no one should be using that emit `DeprecationWarning`

PR #1266 (2020-02-03): Stop setting `logger.propagate=True` on every logger
There are two ways to configure logger handlers in python:

1. Attach a handler to the root logger, and use the fact that records logged to child loggers will automatically inherit this handler by default.
2. Attach a handler to every logger, and disable propagation for every single one.

Up to this commit, we use approach 2, by subclassing `logging.Logger` to use this behavior by default.
The problem with this approach is it makes it very difficult for a user to set their own logging format, as their nested loggers will not propagate as the [... cut]
closed 2020-02-03 by eric-wieser