cocotb/cocotb #1218: Looping with First cause segmentation fault (modelsim)

sfavazza-duagon, 2019-12-02:
Hello, 

I am writing a test-bench to parse an asynchronous signal.
Among the various checks, I need to get whether the monitored signal shows a rising edge, a falling edge or simply never arrive (timeout) transition.
To achieve this I used the `cocotb.triggers.First` method.

When I yield this method in a for loop I always get a segmentation error after the first few iterations.

I attached a simple counter test to reproduce the issue.
I think it is related to the way the scheduler cleans up the not-yet-fired triggers in the `First` method list (I do not understand a lot of the code behind its implementation).

[dummy_counter.zip](https://github.com/cocotb/cocotb/files/3910734/dummy_counter.zip)

eric-wieser, 2019-12-02:
What simulator are you using?

imphil, 2019-12-02:
Title says modelsim

eric-wieser, 2019-12-02:
If I run this test in icarus, I get
```
ValueError: Unable to accurately represent 10(ns) with the simulator precision of 1e0
```

eric-wieser, 2019-12-02:
Once I add the `` `timescale`` directive thought it passes in iverilog:
```
eric@eric-thinkpa:~/win-home/Downloads/dummy_counter$ make
make results.xml
make[1]: Entering directory '/mnt/c/Users/wiese/Downloads/dummy_counter'
PYTHONPATH=/mnt/c/Users/wiese/Downloads/dummy_counter/build/libs/x86_64:/home/eric/win-home/Downloads/dummy_counter:/mnt/c/Users/wiese/Repos/cocotb: LD_LIBRARY_PATH=/mnt/c/Users/wiese/Downloads/dummy_counter/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=dummy_counter_tb \
        TESTCASE= TOPLEVEL=dummy_counter TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /usr/bin/vvp -M /mnt/c/Users/wiese/Downloads/dummy_counter/build/libs/x86_64 -m gpivpi sim_build/sim.vvp -t 10ps
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     Running on Icarus Verilog version 10.1 (stable)
     0.00ns INFO     Python interpreter initialized and cocotb loaded!
     0.00ns INFO     Running tests with cocotb v1.2.0 from /mnt/c/Users/wiese/Repos/cocotb
     0.00ns INFO     Seeding Python random module with 1575284165
     0.00ns INFO     Found test dummy_counter_tb.test_single
     0.00ns INFO     Running test 1/1: test_single
     0.00ns INFO     Starting test: "test_single"
                     Description: Segmentation fault test case
     0.00ns INFO     start the clock
     0.00ns INFO     send a reset pulse
  1000.00ns INFO     RISING EDGE
  1010.00ns INFO     FALLING EDGE
  1020.00ns INFO     RISING EDGE
  1030.00ns INFO     FALLING EDGE
  1040.00ns INFO     RISING EDGE
  1050.00ns INFO     FALLING EDGE
  1060.00ns INFO     RISING EDGE
  1070.00ns INFO     FALLING EDGE
  1080.00ns INFO     RISING EDGE
  1090.00ns INFO     FALLING EDGE
  1091.00ns INFO     Test Passed: test_single
  1091.00ns INFO     Passed 1 tests (0 skipped)
  1091.00ns INFO     **************************************************************************************
                     ** TEST                          PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     **************************************************************************************
                     ** dummy_counter_tb.test_single    PASS         1091.00          0.05     22683.26  **
                     **************************************************************************************

  1091.00ns INFO     *************************************************************************************
                     **                                 ERRORS : 0                                      **
                     *************************************************************************************
                     **                               SIM TIME : 1091.00 NS                             **
                     **                              REAL TIME : 0.07 S                                 **
                     **                        SIM / REAL TIME : 14991.44 NS/S                          **
                     *************************************************************************************

  1091.00ns INFO     Shutting down...
make[1]: Leaving directory '/mnt/c/Users/wiese/Downloads/dummy_counter'
```

eric-wieser, 2019-12-02:
@sfavazza-duagon, Can you show the log output of ` COCOTB_SCHEDULER_DEBUG=1 make`

sfavazza-duagon, 2019-12-02:
Thanks for the prompt response!

I attached both the transcript and the simulation logs.

[sim.log](https://github.com/cocotb/cocotb/files/3911042/sim.log)
[transcript.log](https://github.com/cocotb/cocotb/files/3911043/transcript.log)

Hope this is helpful enough.

sfavazza-duagon, 2019-12-09:
Any news so far?

I would be really happy to help, but I don't know where to start. 
Do you have perhaps a document describing the C-code part of CoCoTb or should I rather dive into the code myself until I get the concerned part?

eric-wieser, 2019-12-09:
I'd recommend trying to start cocotb under gdb, which you might be able to work out how to do from the makefiles alone. I'll try and work it out locally, if you can't.

Unfortunately, without access to modelsim there's not much I can do to help here.

cmarqu, 2019-12-09:
Maybe https://cocotb.readthedocs.io/en/latest/troubleshooting.html is helpful. (Please suggest improvements that you identify during debugging.)

sfavazza-duagon, 2019-12-10:
@eric-wieser in case you've got time, there is the *Intel modelsim - starter edition* (https://fpgasoftware.intel.com/?edition=pro&platform=windows&product=modelsim_ae#tabs-2) simulator which comes at no cost.

@cmarqu thank you for the hint.
Actually I tried in the past to work with the `COCOTB_ATTACH` variable, but in Modelsim the PID is printed in the sim.log file which is unreadable at simulation time.
Only when I quit simulation it prints out the PID in shell, hence I cannot attach a python debugger...

But I'll try with gdb.

cmarqu, 2019-12-11:
> Actually I tried in the past to work with the `COCOTB_ATTACH` variable, but in Modelsim the PID is printed in the sim.log file which is unreadable at simulation time.

Are you saying the whole file is empty until simulation ends?

If you are running from a git checkout, you could try replacing the line 
https://github.com/cocotb/cocotb/blob/a62759901c30db72b592da6c559dbe71efa5b518/cocotb/share/lib/embed/gpi_embed.c#L167
with
```
fprintf(stderr, "Waiting for %lu seconds - attach to PID %d with your debugger\n", sleep_time, getpid());
```
Another option would be to use something like ``pidof vsim`` or ``ps aux | grep vsim``.

sfavazza-duagon, 2020-02-03:
Is there a way to debug the python code once the make command is launched? 
Or in more general terms, a way to attach a pdb to the running python interpreter instance?

Which was your workflow when you developed CoCoTb? I am sure you needed to step through the python code at a certain point...

eric-wieser, 2020-02-03:
> Or in more general terms, a way to attach a pdb to the running python interpreter instance?

Replacing the first line of your test with `import pdb; pdb.set_trace()` is an easy way to acheive that.

sfavazza-duagon, 2020-02-03:
I guess that I got one of those problematic simulators. 
Actually before posting my last question I already tried to place `import pdb; pdb.set_trace()` but the transcript basically stops and hangs there, printing the running processes does not show any python interpreter.

Looking at the `Makefile.questa` I found where the command being executed is created and run.
Yet I do not get where the python interpreter is called, is it somehow called by the auto-generated dll?

eric-wieser, 2020-02-03:
> is it somehow called by the auto-generated dll?

Yes. here:

https://github.com/cocotb/cocotb/blob/8561f137b599a2be2e1100a7ace5a535f7d03350/cocotb/share/lib/embed/gpi_embed.c#L145

eric-wieser, 2020-02-03:
Perhaps try using https://pypi.org/project/remote-pdb/, which should get around that stdin issue.

sfavazza-duagon, 2020-02-03:
> Indeed, the interpreter is part of the modelsim process, it doesn't run separately.

So, basically there is only one PID generated, the one from the simulator, am I right? So there is no point in trying attaching pdb to cocotb/schedule.py.

> Perhaps try using https://pypi.org/project/remote-pdb/, which should get around that stdin issue.

How is a remote python-debugger useful in this context?

eric-wieser, 2020-02-03:
You can launch the remote debugger inside the simulator, and connect to it via localhost from a separate terminal

eric-wieser, 2020-02-03:
> So, basically there is only one PID generated, the one from the simulator, am I right?

Correct

sfavazza-duagon, 2020-02-03:
Should I ever manage to miraculously debug code in python, would it be worth to add the procedure to the `Troubleshooting` section of the CoCoTb guide?

eric-wieser, 2020-02-03:
Absolutely!

sfavazza-duagon, 2020-02-03:
Ok will make a pull request to update the guide.

Btw I managed to connect to a debugger in remote (I report the receipt for your convenience):

1. Install `remote_pdb` python package
2. In the file to be debugged define a remote breakpoint: 
```python
from remote_pdb import RemotePdb
RemotePdb('127.0.0.1', 4444).set_trace()
```
3. open the simulator with `GUI=1`
4. open a terminal and connect through telnet:
```shell
telnet.exe 127.0.0.1 4444
```

At this point you should see a pdb instance being opened.

ktbarrett, 2020-04-22:
Did the original issue get resolved?