cocotb/cocotb #1220: Issue use handle.drivers() in Python 3.7+

ktbarrett, 2019-12-03:
This uses a `next`-like function, but it's defined in C. See it catches `GeneratorExit`.

https://github.com/cocotb/cocotb/blob/17ce4e51da43b6d2d1b5834d629374ed5535a8d8/cocotb/handle.py#L572-L580

The C however throws `StopIteration`. This will bubble up instead of stop the generator.

https://github.com/cocotb/cocotb/blob/17ce4e51da43b6d2d1b5834d629374ed5535a8d8/cocotb/share/lib/simulator/simulatormodule.c#L528-L556

I will make a test for this soon.

eric-wieser, 2019-12-03:
You're right, that looks like nonsense to me.

Introduced in https://github.com/cocotb/cocotb/commit/1719e9875168e1ecb76d0e5e59710576c2b123c7

eric-wieser, 2019-12-07:
Fixed in #1221

closed 2019-12-07 by eric-wieser