cocotb/cocotb #1222: Casts on handles

ktbarrett, 2019-12-03:
Most leaf handle types (`ConstantObject` and `NonConstantObject` and it's children) have cast overloads associated with them which do the convenience behavior of `cast(handle.value)`. This works for `int`, `str`, and `float` on objects that support them. What's missing is `bool`, it was assumed because the previous worked, that a `bool` cast on a VHDL `Boolean` object would as well. Instead it uses the default behavior for bool on a value: `False` if `False`, `0`, `""`, or `None`, and `True` otherwise. Because a handle is non of the false literals, it is always `True`.

I don't think we should add `__bool__`, but instead remove the other interfaces to maintain consistency. The actual operation it offers is trivial. Forcing the users to use `handle.value` instead makes reading and writing to handles more obvious when reviewing source code.

eric-wieser, 2019-12-03:
To be clear, you're suggesting asking the user to change `int(dut.signal)` to `int(dut.signal.value)`?

We definitely shouldn't force users to change, but we could start emitting warnings if they don't.

I suspect this might be really quite intrusive though.

ktbarrett, 2019-12-04:
> To be clear, you're suggesting asking the user to change int(dut.signal) to int(dut.signal.value)?

Yes, and that would be the extent of the change. For `int`, `str` and `float`. 

> I suspect this might be really quite intrusive though.

It likely would be. The alternative is we could add `__bool__`. Like the `int` cast it would fail if the signal was non-0/1 and return `True` if the signal was all '1' and '0' otherwise, just like C's bool cast behavior. Of course, then anyone using the following idiom would silently start doing the wrong thing.

```python
handle = dut.signal if check() else None
# ...
if handle:
    stuff()
```

If we do nothing, this code will silently do the wrong thing.

```python
if bool(dut.fifo.ready):
    dut.fifo.data <= data
    dut.fifo.en <= 1
```

This will never silently do the wrong thing.

```python
if bool(dut.fifo.ready.value):
    dut.fifo.data.value = data
    dut.fifo.en.value = 1

eric-wieser, 2019-12-04:
Note that defining `__bool__` to raise or warn is a very easy way to stop either party from doing the wrong thing silently

ktbarrett, 2019-12-04:
With that solution, cocotb would complain always in the first case, even though that usage is technically correct. The warning could mention "use `if handle is not None:`", but someone has to change their code to not get an earful.

ktbarrett, 2020-01-13:
@eric-wieser I ran into another bug today because of the behavior of the bool cast, I'd like to find a solution. Would you be fine with a definition of `__bool__` being `bool(int(self))` and also raise a warning that the semantics of `__bool__` have changed?

eric-wieser, 2020-01-13:
Typically warnings are used to indicate that something will change soon, not that it has already changed.

> but with any solution someone has to change their code to not get an earful.

Correct, and that's kind of the point - that earful is what motivates users to change their code before it becomes wrong.

I'm tempted to add
```python
def __bool__(self):
    warnings.warn(
        "`bool(handle)` is deprecated. To keep the old behavior of checking if the handle is `None`, use `handle is not None`. If your intent was to check the value of the signal, use `bool(handle.value)` instead. In future, this will be an error.", DeprecationWarning, stacklevel=2)
```

ktbarrett, 2020-01-14:
I'm not a fan of the inconsistency, so I tried several things out. Turns out even internal code `handle.__hasattr__` depends upon the current behavior.

ktbarrett, 2020-04-10:
#1510 fixes the issue from the last comment. Willrevisit this.

closed 2025-02-21 by ktbarrett