cocotb/cocotb #1225: Allow a custom entry point

eric-wieser, 2019-12-04:
Currently, the entry point from C to python is hard-coded here:

https://github.com/cocotb/cocotb/blob/17ce4e51da43b6d2d1b5834d629374ed5535a8d8/cocotb/share/lib/embed/gpi_embed.c#L398

Which is

https://github.com/cocotb/cocotb/blob/ba0fb097efbd2939ad6a961ee8d02793104ef8f7/cocotb/__init__.py#L108

From about here, the things it does become opinionated:

https://github.com/cocotb/cocotb/blob/ba0fb097efbd2939ad6a961ee8d02793104ef8f7/cocotb/__init__.py#L154-L172

It would be great if a user could set an env var or plusarg to launch into a different entry point

ktbarrett, 2019-12-05:
Plusargs don't work with Questa. Don't know why anyone didn't think of that when they added the feature. I'm not a fan of environment variables when we could use a Python module (or modules).

This would require some modification of the scheduler and regression manager. They both reach into eachother by way of global scheduler and regression manager variables. Custom code might not use either.
https://github.com/cocotb/cocotb/blob/17ce4e51da43b6d2d1b5834d629374ed5535a8d8/cocotb/__init__.py#L89-L92

eric-wieser, 2019-12-05:
>  I'm not a fan of environment variables when we could use a Python module (or modules).

There are two approaches to this problem I've seen in the wild:

* `conftest.py` (pytest), `conf.py` (Sphinx) - this isn't really an entry point, but is an example of settings being loaded from a python file. Note that I'm pretty sure this isn't imported the usual way, but is read from the file system and then exec'd, which gives more flexible rules about where it can live.
* a `--entry-point=my.python_module:my_func` syntax, which is how other runtime environments (eg the python WSGI web server hosts, `cherrypy`, `gunicorn`, etc) let you choose an entry point. The dot and colon syntax is a `setuptools` protocol, so seems sensible to reuse.

My preference is towards the latter, because a the world with arbitrary entry points (eg IPython), cocotb feels more like a runtime environment than a configurable test suite.

ktbarrett, 2019-12-05:
>I think @ktbarrett's comments above are in favor of something like the former.

Correct. I don't see a reason that normal imports can't be used. After looking at your examples, I like pytest's approach of calling well defined hooks, but instead of hooks overrides. The overrides can call the default implementation if they just want to hook behavior.

>because a the world with arbitrary entry points (eg IPython), cocotb feels more like a runtime environment than a configurable test suite

And why is that a bad thing? There is a lot of useful technology in cocotb that is difficult to reuse because of how tightly coupled everything is. IMO GPI should be a separate repo entirely and should load into any arbitrary python module. Cocotb could exist as a configurable test suite that uses GPI. At that point an interactive testing framework using IPython would simply be another consumer of the GPI. I think the problem is that people are thinking about cocotb too much as "just another Python package".

eric-wieser, 2019-12-05:
> And why is that a bad thing?

I don't think it is at all! My point was that if we're a runtime environment, the wsgi model is a closer fit than the pytest one.

ktbarrett, 2019-12-05:
Oh sorry, I usually get a lot of resistance at work. I'm projecting. So you're saying push a module and entry function via an environment variable for C to load as an application entry inplace of hardcoding `cocotb` and `_initialize_testbench`? If so, I wholeheartedly agree.

eric-wieser, 2019-12-06:
Yes, that's pretty much what I'm suggesting

imphil, 2019-12-27:
Would you mind outlining an end-user use case for such a functionality? (What problem are we solving?)

ktbarrett, 2019-12-28:
@imphil The first thing that comes to mind is entry into IPython or a JupyterNotebook for interactive testing rather than the regression manager. Think PYNQ, but with a simulator instead of hardware. Perhaps even alternatives to the regression manager, like pytest could be easily prototyped. Doing so would allow developers of those alternatives to develop outside of the main cocotb repo without having to maintain a fork of the entire repo, which is good for all parties. There will need to be some refactoring to give cocotb (the Python framework) a more consistent interface for the GPI to effectively black-box it. A custom entry point is just one step of many.

eric-wieser, 2019-12-28:
@ktbarrett has nailed the motivation there - ipython and pytest are the two most relevant use-cases in my mind.

garmin-mjames, 2020-01-10:
While we're discussing the idea of custom entry points to Python, it's also worth considering decoupling loading GPI from embedding Python.
That allows the use of GPI with testing frameworks in other languages, or using some kind of ipc.

Some discussion of this in #1181.

ktbarrett, 2020-01-12:
@garmin-mjames Thank you for mentioning that. cocotb really should be many libraries IMO. GPI, PyGPI, and any number of Python testing frameworks whether it be the current regression manager, pytest-based, interactive, etc. I don't know if the support for that should be in this repo, or if it should be moved into another repo, or if the whole thing just makes people uncomfortable.

ktbarrett, 2020-12-04:
I have two potential solutions coded up, but not polished.

[The first](https://github.com/ktbarrett/cocotb/tree/custom-entry-1) add the entry point customization at the end of `_initialize_testbench` before creating the scheduler and regression manager, allowing the "common" case of loading into an experimental scheduler/regression system, or adding hooks to those object or modifying the logger before continuing. It is the simpler of the two.

[The second](https://github.com/ktbarrett/cocotb/tree/custom-entry-2) adds the entry point customization to the gpi_embed library, so it can load into a completely different library and sidestep the `cocotb` package altogether. This is the more flexible option, but the bigger change. The gpi_embed calls into it's own Python library (for simplicity) to parse the entry point spec and and return the module and function to gpi_embed. gpi_embed gathers the other bits from the module it needs (hopefully that can be removed in the future, and the user can call register functions) and then calls into the entry point. I also took the liberty of cleaning up the `embed_sim_init` function, since I was having a hard time reasoning about the lifetimes of the objects.

garmin-mjames, 2021-01-12:
I like the second approach better. It's flexible and allows the user to run an application within the simulator environment but not necessarily use the cocotb test suite.

PR #2484 (2021-08-23): Load custom entry point in Python from the GPI
Closes #1225. From https://github.com/cocotb/cocotb/issues/1225#issuecomment-738573856, solution 2. Sorry the diff is mostly unreadable. I found it was more work than it was worth to split the changes between DEFER and the rest.

The value of this PR is that we can easily experiment with alternative regression managers and schedulers while keeping existing code working by loading into another entry point. It can also be used by users to setup hooks on cocotb functionality before a test runs.

gpi_embed is no longer hardcoded to enter
"cocotb:_init [... cut]
closed 2021-08-23 by ktbarrett