cocotb/cocotb #1226: Need a logging level between DEBUG and INFO

jahagirdar, 2019-12-04:
Logging level DEBUG prints lots of information about cocotb internals which is not useful for a testcase writer or a dut debugger.
Usage of Levels above INFO make sense only if we want to alert the user about some potential issue (unimplemented features, test errors)
So for a testcase writer INFO gets crammed with two types of information.

1. General information( e.g. Test state like "Reset Done" "Sending Packet No. X", "Received Packet Y" etc.)
2. Test Debug Information (Packet content was {list of bytes}, Configured register X with value Y etc)
Creating a loglevel between DEBUG and INFO will help in separating these two usecases.

eric-wieser, 2019-12-04:
You can set the level per logger in your test module, outside of any tests. Does that help?

eric-wieser, 2019-12-04:
Perhaps we need some better docs here

cmarqu, 2019-12-04:
Related: #493

jahagirdar, 2019-12-06:
@eric-wieser 

> You can set the level per logger in your test module, outside of any tests. Does that help?

In addition to setting the log level we may require a separate `cocotb.log.debug` and `cocotb.log.core_debug` function

eric-wieser, 2019-12-06:
For logging your own things, i'd recommend:
```
my_logger = logging.getLogger(__name__)
my_logger.setLevel(logging.DEBUG)

# Use my_logging.debug below
```

cmarqu, 2021-04-18:
Some or all of this seems to be solved by #2502 now.

closed 2021-05-26 by github-actions[bot]