cocotb/cocotb #1227: cocotb.config file: Do we need it?

jahagirdar, 2019-12-04:
One of the recommendation's in PR #1224 was to read user configuration from a configuration file instead of creating separate environment/make variable for each case.

I already am using something similar in my test environment and wanted to check whether there is a wider interest in using something like this.

The way I see this being modeled is.

1. There are multiple cocotb.config files
2. Each file has one or more sections (e.g. `default`, `testsuite_name`, `testcase_name`)
3. The code will search for cocotb.config files in predefined hierarchy (cocotb installation, all directories in the path to testcase)
3. For a given config file, It will first apply settings in default, then testsuite the testcase. settings in testsuite will override settings with same name in default. settings in testcase will override settings in testsuite and default.
4. The config files will be applied in the order( cocotb install, root folder, root -1 folder,.... test folder) with declarations in each config file overriding the previous declarations.
5. The final override will be via environment variable.

This can be implemented via a script which is read in by the makefile before invoking the sim/regress target or as a python module within cocotb.

# Advantage:
By changing the settings in one file, I can modify the entire/part of test suite to accommodate new requirements.

# Disadvantage:
For a person who does not know the environment, sudden magic can happen if a team member changes the root config file

Thoughts welcome.

eric-wieser, 2019-12-04:
Are you referring to my first or second bullet point in the comment in the linked issue?

jahagirdar, 2019-12-04:
I was referring to 
```
Add a COCOTB_LOGGING_CONFIG_FILE env var, which if present is used to call logging.config.fileConfig
```
And converting it to a general mechanism for loading all test and cocotb configurations from a set of config file.

ktbarrett, 2019-12-06:
I don't know if we need a multi-section config file for test suites or test cases, we can change the loggers freely once in user-controlled code. The only thing we can't do is affect the global loggers. A single environment variable and single configparser file to load at `__init__` is sufficient for that.

eric-wieser, 2019-12-06:
>The only thing we can't do is affect the global loggers.

I believe this is only true because cocotb sets `propagate = False`, which is not helpful.

If we didn't do that, I think something like
```
logging.config.fileConfig(...)
```
at the top of your test file would work just fine.

ktbarrett, 2019-12-06:
Would there be a performance effect to setting `propagate` to `False`? There are also a number of messages logged before a test module is loaded, an important one being `RANDOM_SEED`.

I think we could add a `setup_logging` function to the `cocotb.log` module that would use `logging.config.fileConfig` on the value of `COCOTB_LOG_CONFIG` if it is present in the environment. I'll put up a draft later.