cocotb/cocotb #1234: argv gets mangled with cadence xcelium

wimiko, 2019-12-10:
I see that ARGV gets mangled when using excelium. cocotb crashes, complaining about invalid utf8 characters.
The curious thing is: it only happens to the argument after the -f switch. I have modified `VpiStartupCbHdl::run_callback()` in VpiCbHdl.cpp to print info.argv and when i have 

`xrun  -v93 -f xrun_files_cmd -timescale 1ns/1ps -licqueue ... `

out comes:
`
xrun
-v93
-f
���
-timescale
1ns/1ps
-licqueue
...
`
If I move the -f option around, the filename stays mangled. If I have multiple -f options, i get multiple garbled filenames. All other options go through fine.

eric-wieser, 2019-12-10:
It might be instructive to print the raw byte values of the garbled argument - perhaps they are being replaced with file numbers

wimiko, 2019-12-13:
So, if I print the characters:

xrun  -v93 -f xrun_files_cmd -timescale 1ns/1ps 

xrun
78 72 75 6e 
-v93
2d 76 39 33 
-f
2d 66 
+M
10 2b 4d 05 
-timescale
2d 74 69 6d 65 73 63 61 6c 65 

The numbers change between runs. 
If I have multiple -f switches, with different files, it looks like this:

xrun
78 72 75 6e 
-v93
2d 76 39 33 
-f
2d 66 
���
ffffffd0 ffffffa0 ffffffea 03 
-f
2d 66 
���
ffffff90 ffffffe8 ffffffec 03 
-input
2d 69 6e 70 75 74 
xrun_commands.tcl
78 72 75 6e 5f 63 6f 6d 6d 61 6e 64 73 2e 74 63 6c 

Interestingly, the argument to the -input switch, which is also a filename comes trough ok.
Since the -f switch provides a file with more arguments, my guess is that the value is a reference to another structure that contains the arguments in the file, or something like that.

Does cocotb actually need these arguments for something? Maybe we can just let it fail in silence?

eric-wieser, 2019-12-20:
> Maybe we can just let it fail in silence?

We certainly could, but it would be cool to understand what the value we're getting instead is.

garmin-mjames, 2020-04-30:
I think I found answer to this.

From the 2001 standard:
> The argument following a -f argument shall contain a pointer to a NULL terminated array of pointers to characters. This new array shall contain the parsed contents of the file. The value in entry zero shall contain the name of the file. The remaining entries shall contain pointers to NULL terminated character arrays containing the different options in the file. The last entry in this array shall be a NULL. If one of the options is a -f then the next pointer shall behave the same as described above.

From the 2005 standard (worded slightly differently):

>  The vendor tool may provide a command-line option to pass a file containing a set of options. In that case, the argument strings returned by vpi_get_vlog_info() shall contain the vendor option string name followed by a pointer to a NULL-terminated array of pointers to characters. This new array shall contain the parsed contents of the file. The value in entry zero shall contain the name of the file. The remaining entries shall contain pointers to NULL-terminated character arrays containing the different options in the file. The last entry in this array shall be NULL. If one of the options is the vendor file option, then the next pointer shall behave the same as described above.

So the number is a pointer to an array of pointers, each one of those a pointer to a string.
Looks like 2001 indicates a '-f' option, but 2005 allows for other options.

eric-wieser, 2020-05-01:
Perhaps we should create a class like
```
class fileopt(str):
    def __new__(cls, fname, argv):
        self = super().__new__(cls, fname)
        self.argv = argv
        return self
```

garmin-mjames, 2020-05-01:
Wouldn't we want to replace the pointer address with the filename, but leave the other options obscured under that instead?

Using the example above `xrun -v93 -f xrun_files_cmd -timescale 1ns/1ps`,
I would think we'd want `xrun_files_cmd` to appear in `cocotb.argv`.

Options:

- Put `-f xrun_files_cmd` in argv and put nested structure under `xrun_files_cmd`
- Expand `argv` by adding the commands from the file after `-f xrun_files_cmd`
- Replace `-f xrun_files_cmd` with the contents of the file

eric-wieser, 2020-05-01:
Under my proposal, `argv[n] == fname`, `argv[n].argv = ['flags', 'in', 'file']`, recursively if needed