cocotb/cocotb #1245: Questa, ModelSim: ReadWrite after ReadOnly

ktbarrett, 2019-12-19:
As discovered in #834, the test `test_readwrite_in_readonly` is expected to fail for "ModelSim" simulators, but when using Questa 10.5c and 10.7c the test passes. The test checks if the following fails:

```python
@cocotb.coroutine
def do_test_readwrite_in_readonly(dut):
    yield RisingEdge(dut.clk)
    yield ReadOnly()
    yield ReadWrite()
```

Why is this expected behavior? Why do the other simulators listed fail? Does Questa <10.5 or ModelSim actually fail here?

eric-wieser, 2019-12-20:
Relevant lines:

https://github.com/cocotb/cocotb/blob/6d4ce2593bb720f01a9fd6b1287bc99ab9d1957f/tests/test_cases/test_cocotb/test_cocotb.py#L348-L354

https://github.com/cocotb/cocotb/blob/6d4ce2593bb720f01a9fd6b1287bc99ab9d1957f/tests/test_cases/test_cocotb/test_cocotb.py#L375-L390

eric-wieser, 2019-12-20:
I'm not sure I understand why this is marked both as expected failure and expect error.

ktbarrett, 2019-12-22:
@eric-wieser I'm guessing it's because it causes some simulators to fail hard (`SimFailure`) and others to not re-enter (the coro will not continue to run `exited = True`) and will fail because of the time via `TestFailure`. The `expect_fail` case should also be predicated on the simulator, but all simulators apparently failed the test, so `True` was an easy catch-all.

eric-wieser, 2019-12-25:
I assume the expectation here is that the user must wait for `NextTimeStep` before waiting for `ReadWrite` again?

While obviously we should get the tests to pass, we should perhaps either make this fail on quests artificially, or we should make the other simulators work.

ktbarrett, 2019-12-27:
I'm in favor the former, we could also get all the failing simulators to error instead. Silently not working is bad.

ktbarrett, 2020-09-08:
GHDL and Verilator now also fail on this test.

PR #4208 (2024-10-09): Prevent transitions from ReadOnly to ReadWrite or ReadOnly
Transitioning from ReadOnly to ReadWrite or ReadOnly are not legal, this checks prevents that from occuring by raising a `RuntimeError` when it is attempted. Closes #1245. Closes #2222.
closed 2024-10-09 by ktbarrett