cocotb/cocotb #1246: Support forking non-decorated async coroutines

ktbarrett, 2019-12-19:
Support the following example of forking an non-decorated async coroutine.

```python
async def example():
    for i in range(10):
        await cocotb.triggers.Timer(10, 'ns')

cocotb.fork(example())
```

ktbarrett, 2019-12-19:
One can use `coro_obj.__await__()` to obtain a generator from an async coroutine. You can obtain `__name__` and `__doc__` from the async coroutine object as expected. However, you can't obtain the module the coroutine function was defined in directly, you must do some math on `coro_obj.cr_code.co_filename`. And I have a feeling this is an implementation detail that will eventually change.

eric-wieser, 2019-12-20:
To get hold of generator objects:
```
e = example()
gen = e.__await__()  # generator-like thing
name = e.__name__
```
As mentioned in a now-deleted comment, it's not clear how to get the module name - but I think that's ok.

ktbarrett, 2019-12-21:
@eric-wieser My solution is just to get rid of the module name, etc. code from `RunningCoroutine`, it's only used by `RunningTest` anyways, which will always be decorated.

PR #1255 (2020-02-02): Support non-decorated async coroutine in fork
Resolves #1246. `scheduler.add` accepts non-decorated async coroutines.
closed 2020-02-02 by imphil