cocotb/cocotb #1253: Tests that fail before the scheduler are not handled properly

ktbarrett, 2019-12-22:
For example:
https://github.com/cocotb/cocotb/blob/92a032a4349b2fe9e2f9bff5d1060ef44cd4cd72/tests/test_cases/test_cocotb/test_cocotb.py#L85-L88

This is considered "N/A" in the regression results banner and passes whether `expect_fail` or `expect_error` are set to `True` or `False`. A traceback is printed, but it's not apparent there was any failure if you are running a number of tests, like in `test_cocotb`.

The regression manager nor the `cocotb.test` decorator make any attempt to handle potential issues.
https://github.com/cocotb/cocotb/blob/92a032a4349b2fe9e2f9bff5d1060ef44cd4cd72/cocotb/regression.py#L157
https://github.com/cocotb/cocotb/blob/92a032a4349b2fe9e2f9bff5d1060ef44cd4cd72/cocotb/decorators.py#L477-L478

I put some time into refactoring `initialize` and `handle_result` in the regression manager, but felt like I more or less had to copy-paste to victory, which I didn't like. I think a larger refactor is in order.

ktbarrett, 2019-12-22:
Note, this isn't an issue if the test coroutine is an async function. When async functions are called, nothing executes immediately, so they cannot fail until they get to the scheduler.

eric-wieser, 2019-12-22:
Pretty sure we could detect this with `insoect.isgeneratorfunction` or similar

ktbarrett, 2019-12-22:
I just didn't find a clean way of modifying `handle_result` to call it from `initialize` in the error case, but I see it now with fresh eyes.

eric-wieser, 2019-12-22:
PR appreciated, this has been annoying me for a while when writing and screwing up new tests.

ktbarrett, 2019-12-24:
I'm actually wondering if instead of modifying regression I should modify the decorators so that `__call__` does not immediately run the coroutine's body to the first `yield`. async coroutine functions have that deferred execution behavior, it doesn't make much sense for async and non-async coroutines to have different behavior.

eric-wieser, 2019-12-24:
> The issue is with errors that cause coroutine initialization to fail like NameError, not necessarily any exception before the first yield.

I suspect the issue is actually errors during coroutine initialization, such as:
```
@cocotb.coroutine
def not_a_coroutine(dut):
    raise RuntimeError()
```
or
```
@cocotb.coroutine
def returns_a_coroutine(dut):
    return the_coroutine_with_a_typo()
```

PR #1262 (2020-02-07): Do not ignore failures during test initialization
Fixes #1253. PEP8 fixes and comments.

There are two places in the RM that initialize coroutines, each had slightly different code, so this was factored out into a method `init_test`.

`init_test` can fail, and so can a test at any point in it's run (previous handled by `handle_result`), so scoring a test as a pass or a fail depending upon the result and expected behavior is factored out into a method `score_test`.

`score_test` can print out tracebacks on certain failures, and non-pertinent traceback frames are stripped. However because there  [... cut]
commit ca9a45d2 (2020-02-07): Do not ignore failures during test initialization. Fixes #1253.

There are two places in the RM that initialize coroutines, each had
slightly different code, so this was factored out into a method
init_test.

init_test can fail, and so can a test at any point in it's run
(previously handled by handle_result), so scoring a test as a pass or a
fail depending upon the result and expected behavior is factored out
into a method score_test.

Now results are handled during initialization, which fixes #1253.
closed 2020-02-07 by eric-wieser