cocotb/cocotb #1261: Simple Python exception causes segmentation fault on vcs

markmelvin, 2019-12-24:
We have been struggling with random segmentation faults with our verification suite and VCS and until now we had assumed it was a problem with our code.  However, recently I have noticed similar behaviour with a very simple test case which appears to be a problem with cocotb itself.

Let's say I have the following simple test:

```python
import cocotb

@cocotb.test()
def check_top_level(dut):
    """Checks the simulation was built and can be clocked."""
    yield do_something(dut)
```

Where in another file I have this:
```python
import cocotb
# Note I have commented out this import on purpose...
# from cocotb.triggers import Timer

@cocotb.coroutine
def simple_coroutine(dut):
    yield Timer(500)

@cocotb.coroutine
def do_something(dut):
    # dut.i_bank_rst__b_a = 0    # Uncomment me to cause a segfault
    yield simple_coroutine(dut)

```

Now, if you run this as-is you will get an exception due to Timer not being resolved.  However, if you access any of the verilog itself (e.g uncomment the line that resets a reset signal in our design), then what would normally be thrown as an exception now causes a segfault.  This seems to be an issue with the embedded Python interpreter shutting down before the exception is printed and then Python tries to access invalid memory and causes a segfault (at least that is our guess).  

We have seen similar issues (segfault or a lockup) happen when a Python exception happens after we have accessed the verilog, but before the first yield of a coroutine.  It seems to be related to the embedded Python in cocotb doing something wrong when shutting down.

I think if we could get to the root of this issue in cocotb, we could then begin to get to the real root of our segfaults - which could be a simple Python exception coming from somewhere.

ktbarrett, 2019-12-24:
Are you using the most recent version of cocotb? We have struggled with the same issue using Questa for some time, but I have not seen it recently.

Also, this is related to #1253. The test is failing before the test coroutine makes it to the scheduler. And there is no error handling at all if a test should fail in this way.

A work around for now is to insert a `cocotb.triggers.NullTrigger()` as the first statement in the test, or make the test an async coroutine:

```python
@cocotb.test()
def check_top_level(dut):
    """Checks the simulation was built and can be clocked."""
    yield cocotb.triggers.NullTrigger()
    yield do_something(dut)
```

```python
@cocotb.test()
async def check_top_level(dut):
    """Checks the simulation was built and can be clocked."""
    await do_something(dut)
```

markmelvin, 2019-12-24:
This is with the latest version.  Maybe I am mixing two issues here.  We have also worked around the potential lockup and some segfaulting issues by wrapping cocotb.coroutine with an exception-handling coroutine that always yields a dummy Timer no matter what happens.  I guess it is similar to your workaround - but I can see if that gets rid of my segfault I can reproduce now.

markmelvin, 2019-12-24:
Hi @ktbarrett - I tried your suggestion of yielding a NullTrigger() first and that did not do anything.  Segfault still occurs.  It is a bit annoying because the segfaulting process keeps trying to re-launch itself. I'll see if I can debug this further to find out where it is actually segfaulting.

eric-wieser, 2019-12-24:
Can you show the segfault output with PYTHONFAULTHANDLER set?

markmelvin, 2019-12-24:
Yeah, it gives no info.  It looks like it is probably segfaulting in vcs...probably in the code that accesses the dut??

The current stacktrace is:
```*** Error in `sim_build/simv': munmap_chunk(): invalid pointer: 0x0000000000d5bbf0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7f754)[0x2aaab2b7f754]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(+0x25e9b)[0x2aaaabf69e9b]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(+0x1f5cc)[0x2aaaabf635cc]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(+0x1be6a)[0x2aaaabf5fe6a]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(+0x1c29d)[0x2aaaabf6029d]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(_ZN3vfs22cleanupVcsDbFileSystemEv+0x9)[0x2aaaabf67349]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so(vfs_cleanup+0x9)[0x2aaaabf690c9]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(VCS_EXIT+0x65)[0x2aaaaf2c4845]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x1cf91c5)[0x2aaaadf4a1c5]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(SigHandler+0x549)[0x2aaaadd91dd9]
/lib64/libpthread.so.0(+0xf5d0)[0x2aaab22ea5d0]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x10dce71)[0x2aaaad32de71]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x11573af)[0x2aaaad3a83af]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x115d74b)[0x2aaaad3ae74b]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x114d25f)[0x2aaaad39e25f]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(LastEventRoutineTCMP32NOTC+0xd99)[0x2aaaadeb6489]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x1c9334d)[0x2aaaadee434d]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(+0x1d0cebf)[0x2aaaadf5debf]
/server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so(VCS_MAIN+0x13b4)[0x2aaaadf741f4]
sim_build/simv[0x40c807]
```

markmelvin, 2019-12-24:
Got a core dump - a little more info:

```
Core was generated by `sim_build/simv +define+COCOTB_SIM=1 +vcs+grwavesoff +vpdignore +vcs+dumpvarsoff'.
Program terminated with signal 11, Segmentation fault.
#0  0x00002aaab2b362c7 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install bzip2-libs-1.0.6-13.el7.x86_64 dbus-glib-0.100-7.el7.x86_64 dbus-libs-1.10.24-13.el7_6.x86_64 elfutils-libelf-0.172-2.el7.x86_64 elfutils-libs-0.172-2.el7.x86_64 expat-2.1.0-10.el7_3.x86_64 glib2-2.56.1-4.el7_6.x86_64 glibc-2.17-260.el7_6.6.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.15.1-37.el7_6.x86_64 libattr-2.4.46-13.el7.x86_64 libblkid-2.23.2-59.el7_6.1.x86_64 libcap-2.22-9.el7.x86_64 libcom_err-1.42.9-13.el7.x86_64 libffi-3.0.13-18.el7.x86_64 libgcc-4.8.5-36.el7_6.2.x86_64 libgcrypt-1.5.3-14.el7.x86_64 libgpg-error-1.12-3.el7.x86_64 libmount-2.23.2-59.el7_6.1.x86_64 libselinux-2.5-14.1.el7.x86_64 libstdc++-4.8.5-36.el7_6.2.x86_64 libuuid-2.23.2-59.el7_6.1.x86_64 lz4-1.7.5-2.el7.x86_64 nss-softokn-freebl-3.36.0-5.el7_5.x86_64 openssl-libs-1.0.2k-16.el7_6.1.x86_64 pcre-8.32-17.el7.x86_64 sssd-client-1.16.2-13.el7_6.8.x86_64 systemd-libs-219-62.el7_6.7.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-18.el7.x86_64
(gdb) list
1	/local/it/python/Python-3.7.3/Modules/mmapmodule.c: No such file or directory.
(gdb) bt
#0  0x00002aaab2b362c7 in raise () from /lib64/libc.so.6
#1  0x00002aaab2b379b8 in abort () from /lib64/libc.so.6
#2  0x00002aaab2b78e17 in __libc_message () from /lib64/libc.so.6
#3  0x00002aaab2b7f754 in malloc_printerr () from /lib64/libc.so.6
#4  0x00002aaaabf632eb in SNPSle_a696c70aec7e82d9ec9a32a27b62eb4bf61a73985d2edecb () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so
#5  0x00002aaaabf5fe6a in SNPSle_4d2564044a1655d135fc93d41eff971a6534879c7ea6fb33632eed5d8f47837f90388eab8f0eca4a141520287ef8775d () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so
#6  0x00002aaaabf6029d in SNPSle_a696c70aec7e82d940ae9d24ad540f7fcc5f54118c19a41d7034bbf2d1c985ba () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so
#7  0x00002aaaabf67349 in SNPSle_4d2564044a1655d194eb0cd2abe56604fc68ac701a8bbd28c25f2c277f8758f0a84ac13b4a93ace2 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so
#8  0x00002aaaabf690c9 in SNPSle_97c59cdbe8d7a38aee09a8ab955669a3 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvfs.so
#9  0x00002aaaaf2c4845 in SNPSle_946ebca490adacba () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#10 0x00002aaaadf4a1c5 in SNPSle_64133461705005bb725549e2e6fa1b3f () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#11 0x00002aaaadd91dd9 in SNPSle_82244d58c54c18c70d63edc9becab634 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#12 <signal handler called>
#13 0x00002aaaad32de71 in SNPSle_865de9ea592a9a05a5abc2b673acd04e15f83b862807ea7e742f28a9cfeccaa5 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#14 0x00002aaaad3a83af in SNPSle_3ba47e250f393ca0795a177eb4a04fc85f01a55a0c7c15a95a2f02d413260776 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#15 0x00002aaaad3ae74b in SNPSle_06c0eea346e615d974e60e67bdce32dda719f96f498ffe1b () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#16 0x00002aaaad39e25f in SNPSle_c65951064c9d8ed9858c00b76bb59352505120e403bceef31c5e0ed1bb0b425b92bc75a2a6817275 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#17 0x00002aaaadeb6489 in SNPSle_3f1ffb4774cc90f121afbae63023c6e6a5a441ab981992e3c6adc4b703efe533 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#18 0x00002aaaadee434d in SNPSle_63c7abb936d47b07d29d61775f9dbe8c () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#19 0x00002aaaadf5debf in SNPSle_490598bfebcc8e81efcf3d1815066cc9 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#20 0x00002aaaadf741f4 in SNPSle_490598bfebcc8e81 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#21 0x000000000040c807 in main ()
```

I also get these warnings when building.  Could they be related?
```
Warning-[SYSTF-OVERRIDDEN] System task/function call overridden
  System task/function call '$info' is overridden by user defined PLI call 
  through vpi_register_systf. This will supersede the default behavior of the 
  system task/function call
  Please rename the function registered through vpi_register_systf.

Warning-[SYSTF-OVERRIDDEN] System task/function call overridden
  System task/function call '$warning' is overridden by user defined PLI call 
  through vpi_register_systf. This will supersede the default behavior of the 
  system task/function call
  Please rename the function registered through vpi_register_systf.

Warning-[SYSTF-OVERRIDDEN] System task/function call overridden
  System task/function call '$error' is overridden by user defined PLI call 
  through vpi_register_systf. This will supersede the default behavior of the 
  system task/function call
  Please rename the function registered through vpi_register_systf.

Warning-[SYSTF-OVERRIDDEN] System task/function call overridden
  System task/function call '$fatal' is overridden by user defined PLI call 
  through vpi_register_systf. This will supersede the default behavior of the 
  system task/function call
  Please rename the function registered through vpi_register_systf.

[... cut ...]
  The VPI function 'vpi_chk_error' is not supported in the compiler.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.

     -.--ns ERROR    cocotb.gpi                                 VpiCbHdl.cpp:74   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbEndOfSimulation(12)

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.
```

markmelvin, 2019-12-24:
Setting `DEBUG=1` gives a bit more info for this particular issue:

```
--- Stack trace follows:

Dumping VCS Annotated Stack:
#0  0x00002aaab2cdebbc in waitpid () from /lib64/libc.so.6
#1  0x00002aaab2c5cea2 in do_system () from /lib64/libc.so.6
#2  0x00002aaaabacba4c in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313e639d2928478c79b3f673f99203bcf8be64600612100082236bffb2007f1e0ef9 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/liberrorinf.so
#3  0x00002aaaabacd58e in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313efba706aab251478fa49e66610e453774633a6c152e7ef778f2202cda681f3d4e () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/liberrorinf.so
#4  0x00002aaaabac5fc3 in SNPSle_d35ca1ff70d465c2b9b1a72eee90a506fdd009d3de3db1de () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/liberrorinf.so
#5  0x00002aaaae06416f in SNPSle_64133461705005bb725549e2e6fa1b3f () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#6  0x00002aaaadeabdd9 in SNPSle_82244d58c54c18c70d63edc9becab634 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#8  0x00002aaabf3c91a2 in update_refs (containers=0x2aaabf74b5d0 <_PyRuntime+400>) at Modules/gcmodule.c:243
#10 0x00002aaabf3ca8e5 in collect_with_callback (generation=2) at Modules/gcmodule.c:1028
#12 0x00002aaabf3ca94d in _PyGC_CollectIfEnabled () at Modules/gcmodule.c:1587
#13 0x00002aaabf39abeb in Py_FinalizeEx () at Python/pylifecycle.c:1185
#14 0x00002aaabf9538cb in embed_sim_cleanup () at gpi_embed.c:199
#15 0x00002aaabefe1108 in gpi_cleanup () at GpiCommon.cpp:145
#16 0x00002aaabefe10c8 in gpi_embed_end () at GpiCommon.cpp:134
#17 0x00002aaabebc46e5 in VpiShutdownCbHdl::run_callback (this=0xdb5b50) at VpiCbHdl.cpp:474
#18 0x00002aaabebc04de in handle_vpi_callback (cb_data=0xd629e0) at VpiImpl.cpp:556
#19 0x00002aaaad4c08c0 in SNPSle_27959203ce5ad0720e45d4754d3a515910cb8b7ca4bb47195752011ce913284754e716cad71ddff5 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#20 0x00002aaaad4c34f6 in SNPSle_eed38e464c7100f70d40bb073590aafa () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#21 0x00002aaaaf514215 in SNPSle_294d108d9c21223f31bca47d22167e39 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#22 0x00002aaaae0779fd in SNPSle_85eeb05ebf66d014 () from /server/cad/synopsys/vcs-mx/O-2018.09-1/linux64/lib/libvcsnew.so
#23 0x0000000000000000 in ?? ()
Completed context dump phase data

During a VPI callback function for callback reason="cbEndOfSimulation" 
During Error message function Code:="VFS_SDB_ERROR
```

markmelvin, 2019-12-24:
Actually ignore that previous thing.  I believe that error happened due to me trying to attach with a debugger.  That is probably not the root cause.

eric-wieser, 2019-12-28:
The above problem is still real though, right - even if its not the one you were originally debugging?

markmelvin, 2019-12-28:
> The above problem is still real though, right - even if its not the one you were originally debugging?

Perhaps. This occurred when the sim process was relaunching itself after a segfault and I was still trying to connect a debugger to the original process.

markmelvin, 2019-12-31:
OK, I just experienced an exception in our code (seen with Icarus) that undoubtedly would have triggered a segfault in VCS. The problem is when VCS segfaults, I believe it is segfaulting when trying to actually print the original Python exception in a manner very similar to the mechanism outlined in this bug report (and thus the original exception is swallowed/lost).  So if we could get this segfaulting issue fixed (or at least work around it so the Python exception is logged or printed before the segfault occurs), I believe we could get to the bottom of our own hard-to-pin-down bug.

I'm kinda stuck though and I'm not sure how to debug this further.  Any help/suggestions would be appreciated.

eric-wieser, 2019-12-31:
@markmelvin: if you're desperate to work out what your exception is, add some logging to `cocotb.outcomes.Error.__init__` in your local installation, which will definitely be reached if any error occurs

markmelvin, 2020-01-02:
@eric-wieser - thanks, but this didn't work.  I can see my logging info printed for normal test failures, but when a segfault occurs there is no additional information. :(

ktbarrett, 2020-01-03:
I have a shot in the dark for you.

1. The test fails
2. The regression manager calls `tear_down`
3. which calls `simulator.stop_simulator`
4. which calls `gpi_sim_end`
5. which calls `VpiImpl::sim_end`
6. which calls `vpi_control(vpiFinish)`
I think on most simulators this is when the sim ends as `stop_simulator` never returns on Questa or Icarus. But Py_Finalize isn't called until the shutdown callback is run. Can you put in some debugging prints to ensure it is being run?

markmelvin, 2020-01-03:
Thanks @ktbarrett . So some interesting behaviour...if I reproduce the test case as I originally posted in this issue, my debug prints in `cocotb.outcomes.Error.__init__` do actually get printed before the segfault (where they don't with our other rogue crash scenario).  I will try to reproduce the other rogue segfault we are getting with debugging prints inserted to see if Py_Finalize is actually being run.

markmelvin, 2020-01-03:
OK, it seems that events with vcs are just not happening at all.  I put in a ton of prints in `gpi_embed.c`, `GpiCommon.cpp` and `VpiImpl.cpp` and what I see is the following:

* I see my debug messages in `embed_sim_init` (so I know my changes are being compiled in)
* I don't see ANY print statements at all from `GpiCommon::gpi_embed_event`
* I see the first message printed out below right before the end of the sim (right before vcs spits out its little "V C S   S i m u l a t i o n   R e p o r t" , but not the second (in `GpiCommon.cpp`):

```c
void gpi_sim_end()
{
    printf("cocotb: in gpi_sim_end()!!");
    registered_impls[0]->sim_end();
    printf("cocotb: leaving gpi_sim_end()");
}
```

* I never see any calls to `gpi_cleanup` and hence no calls to `embed_sim_cleanup`, which is what ultimately calls `Py_Finalize`.  

When running under Icarus I do see `embed_sim_cleanup` called properly at the end of a simulation.

markmelvin, 2020-01-04:
Interesting... on a hunch I commented out this line:

`registered_impls[0]->sim_end();`

And I see everything proceed, eventually calling `embed_sim_cleanup` through to a full graceful shutdown.  So it seems the VPI implementation of `sim_end` is silently killing everything?

markmelvin, 2020-01-06:
Indeed, running my original test case from this issue, I can now see a proper shutdown and error message with `registered_impls[0]->sim_end();` commented out as per above.  I will try to step through the code and see what exactly is going wrong here and post more info as I get it.

ktbarrett, 2020-01-06:
>I never see this method entered: `VpiImpl::sim_end`
>`registered_impls[0]->sim_end();`
These are one in the same. More than likely the prints in `sim_end` are simply not flushing. `vpiControl(vpiFinish)` in `VpiImpl::sim_end` do not cause an immediate test end on Icarus, but maybe they do on VCS. It might be a good idea to conditionally compile that statement out. What does Icarus do if the `vpiFinish` command is not run?

eric-wieser, 2020-01-06:
> These are one in the same.

Unless `registered_impls[0] == nullptr`, in which case they're not and you get a segfault. Could that be happening?

ktbarrett, 2020-01-06:
Also, if calling `vpiFinish` runs the shutdown callback immediately, `Py_Finalize` *is* called, is the stack destroyed at that point? Does the stack unravel causing `Py_BuildValue` in `stop_simulator` fail? No doubt attempting to return to Python will fail...

markmelvin, 2020-01-07:
With this change, the real segfault I am trying to get to the bottom of still occurs and does not allow cocotb to shut down properly.  So I am clearly wrestling with a more fundamental problem on that one... :(

markmelvin, 2020-01-07:
I have confirmed that simply checking for NULL fixes the original bug in this issue:
i.e.
```C
void gpi_sim_end()
{
    if (registered_impls[0] != NULL)
    {
        registered_impls[0]->sim_end();
    }
}
```

My other segfaulting issue is more fundamental appears to not be cocotb's issue. In our case we are calling into the JVM as part of the simulation and something occasionally goes horribly wrong and takes down everything. :(

markmelvin, 2020-01-09:
> I have confirmed that simply checking for NULL fixes the original bug in this issue:
> i.e.
> 
> ```c
> void gpi_sim_end()
> {
>     if (registered_impls[0] != NULL)
>     {
>         registered_impls[0]->sim_end();
>     }
> }
> ```
> 

Could we get the above fix into version 1.3.0?  Do you need me to make a PR for this?

markmelvin, 2020-01-09:
Woops - looks like I just missed v1.3.0. :(

ktbarrett, 2020-01-10:
@markmelvin The line in question should not need to be conditioned. It's worth investigating why it is in your case, and fix that issue instead.

eric-wieser, 2020-01-10:
I'm struggling to see how `registered_impls` could be null, nothing seems to clear it.

ktbarrett, 2020-01-10:
I would say that it starts out life `NULL` because it's a .bss variable, and never gets set if `register_embed` is never run. But nothing would work if that were the case.

Going back to how he described the original problem, maybe it's VCS memory corruption issue? He is writing 0 to a sim handle, perhaps VCS is losing track of it's handles and writing in the wrong place.

@markmelvin can you run GDB and set a watch on the memory location of `vpi_table`? It should be set in `register_embed` and no where else.

eric-wieser, 2020-01-10:
Can we check `registered_impls.size()`? Perhaps we're indexing beyond the array somehow.

markmelvin, 2020-01-14:
I'll look into this further when I get some time. Hopefully this week.

garmin-mjames, 2020-01-17:
> I think on most simulators this is when the sim ends as `stop_simulator` never returns on Questa or Icarus. But Py_Finalize isn't called until the shutdown callback is run. Can you put in some debugging prints to ensure it is being run?

Icarus and Aldec Active-HDL return from `gpi_sim_end`, which is why I added the `sim_ending` flag in `simulatormodule.c` and set it in `stop_simulator`.
Once the cocotb scheduler returns back into `handle_gpi_callback`, `gpi_sim_end` is called which eventually calls `Py_Finalize`.

markmelvin, 2020-01-20:
How can I build the sim with debug symbols?  Currently `gdb` bails with this error:

`sim_build/simv...(no debugging symbols found)...done`

markmelvin, 2020-01-20:
> Can you print `registered_impls.size()`? Perhaps we're indexing beyond the array somehow.

`registered_impls.size()` is 1

cmarqu, 2020-01-26:
> How can I build the sim with debug symbols?

In theory, I think it should be by setting ``COMPILE_ARGS`` (for building ``simv``) and ( since very recently with #1137) ``CFLAGS/CXXFLAGS`` to ``-g`` or (also?) ``-ggdb``, but I haven't had luck with that yet.

cmarqu, 2020-01-27:
> How can I build the sim with debug symbols? Currently `gdb` bails with this error:
> 
> `sim_build/simv...(no debugging symbols found)...done`

Actually, this may be a red herring. When you do an ``ldd simv`` you'll see that it doesn't list ``libvpi.so`` but it will be loaded later anyway. 
Doing
```
(gdb) b gpi_sim_end
Function "gpi_sim_end" not defined.
Make breakpoint pending on future shared library load? (y or [n]) 
```
and running the program nicely stops at ``gpi_sim_end`` and you can inspect everything and step through.

themperek, 2020-01-27:
@cmarqu Do we have debug enabled all the time: https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/Makefile.inc#L91

cmarqu, 2020-01-28:
@themperek Indeed, and since, like, forever: https://github.com/cocotb/cocotb/commit/71e922b6a6b20141d579882b1a89d871f88120d9

eric-wieser, 2020-03-19:
> > Can you print `registered_impls.size()`? Perhaps we're indexing beyond the array somehow.
> 
> `registered_impls.size()` is 1

And yet `registered_impls[0] == NULL`? The only way that could happen is if `gpi_register_impl(NULL)` is called.

However, there are only 3 places that call this function:
```
C:\Users\wiese\Repos\cocotb\cocotb\share\lib\fli\FliImpl.cpp:
 1033  {
 1034      fli_table = new FliImpl("FLI");
 1035:     gpi_register_impl(fli_table);
 1036      gpi_load_extra_libs();
 1037  }

C:\Users\wiese\Repos\cocotb\cocotb\share\lib\vhpi\VhpiImpl.cpp:
  947  {
  948      vhpi_table = new VhpiImpl("VHPI");
  949:     gpi_register_impl(vhpi_table);
  950      gpi_load_extra_libs();
  951  }

C:\Users\wiese\Repos\cocotb\cocotb\share\lib\vpi\VpiImpl.cpp:
  579  {
  580      vpi_table = new VpiImpl("VPI");
  581:     gpi_register_impl(vpi_table);
  582      gpi_load_extra_libs();
  583  }
```
The C++ standard guarantees that `new` return a non-null value. It's possible that somehow VCS is non-compliant, but seems unlikely.

As a sanity check, can you stick a breakpoint in `gpi_register_impl` and verify the argument is not null?

hashhsah, 2020-05-04:
@markmelvin Could you try adding something like `yield RisingEdge(tb.CLK)` for each clock, at the end of each testcase?

Some how this fixes the problem for me.  Want to verify if this is a valid workaround, before probing further on a possible fix.

hofstee, 2020-05-15:
Also hitting this problem. Adding an `await RisingEdge(dut.clk)` to the end of my test does in fact fix the problem for me.

I would try to strace this but it seems like my version of Docker doesn't support it 🙃

Alex-Mann, 2020-07-11:
I was hitting the same issue with VCS and can confirm that either of the following fixes the issue if put at the end of the sequence:
```
await RisingEdge(tb.clk)
await Timer(1, units='ns')
```

xfguo, 2020-10-05:
same to me when `await Timer(xxx, units='ns')` at the end of the sequence.

JeffChanxf, 2021-08-12:
> I was hitting the same issue with VCS and can confirm that either of the following fixes the issue if put at the end of the sequence:
> 
> ```
> await RisingEdge(tb.clk)
> await Timer(1, units='ns')
> ```

Truly!  The "await RisingEdge(tb.clk)" works just fine for me, with cocotb1.5.2 and VCS (P-2019.06-SP2-1). I revised some components with cocotb updated from 1.0 to 1.5.2, during which all the testcases worked perfectly and smoothly with icarus, while some may encounter the beforementioned segfault.
[image]

mperov, 2022-03-20:
Hello everyone! I think that I could solve problem like this! So I described solution here - https://github.com/mperov/fixSegfaultVCS

Febbe, 2024-02-27:
@linyx94 found out [in this issue](https://github.com/pyuvm/pyuvm/issues/144), that "VCS was unable to generate simulation results and waveforms because at the time VCS tried to output these results, the cocotb simulation program had already been completed and its resources had been released."