cocotb/cocotb #1272: Accessing signal in VHDL module from verilog top

Fatsie, 2019-12-31:
This may be related to #1004 and #1105 but there signal was accessed in verilog module from VHDL top. My problem may also be missing feature due to use of 32 bit Modelsim delivered with Quartus software.

_dff.vhdl_:
```VHDL
library ieee;
use ieee.std_logic_1164.all;

entity dff is
  port(clk: in std_logic; d: in std_logic; q: out std_logic);
end entity dff;

architecture rtl of dff is
begin
  process(clk)
  begin
    if rising_edge(clk) then
      q <= d;
    end if;
  end process;
end architecture rtl;
```
_top.v_:
```verilog
module top(clk, d, q);

input clk, d;
output q;

dff dff(.clk(clk), .q(q), .d(d));

endmodule // top
```
_test.py_:
```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test01_access(dut):
    """
    Segfaulting modelsim simulation
    """
    dut.dff._discover_all()
    yield Timer(1)
```
_Makefile_:
```makefile
CURDIR=$(realpath .)

TOPLEVEL := top

COCOTBMAKEDIR=$(shell cocotb-config --makefiles)

VHDL_SOURCES := $(CURDIR)/dff.vhdl
VERILOG_SOURCES := $(CURDIR)/top.v

export COCOTB_LOG_LEVEL=DEBUG
TOPLEVEL_LANG=verilog
MODULE=test
SIM=modelsim
ARCH=i686
VCOM_ARGS=-2008
WAVES := 1

include $(COCOTBMAKEDIR)/Makefile.inc
include $(COCOTBMAKEDIR)/Makefile.sim
```
Running it (second time so no compiling is done):
```console
(cocotb32_37) [verhaegs@localhost cocotb_modunderscore]$ make
[... 108 lines cut ...]
    Segfaulting modelsim simulation
    """
    dut.dff._discover_all()
    yield Timer(1)

t: "test01_access"
#                                                                                                                      Description: 
#                                                                                                                          Segfaulting modelsim simulation
#                                                                                                                          
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:275  in __gpi_get_handle_by_name        Searching for dff
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:288  in __gpi_get_handle_by_name        Checking if dff is native through implementation VPI
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:209  in create_gpi_obj_from_handle      VPI: Not able to map type vpiVHDLArchitecture(1020) to object.
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:281  in native_check_create             Unable to fetch object top.dff
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:288  in __gpi_get_handle_by_name        Checking if dff is native through implementation FLI
#      0.00ns DEBUG    cocotb.gpi                                  FliImpl.cpp:278  in native_check_create             Looking for child dff from top
# ** Error (suppressible): (vsim-FLI-3155) The FLI is not enabled in this version of ModelSim.
# ** Error (suppressible): (vsim-FLI-3155) The FLI is not enabled in this version of ModelSim.
# ** Error (suppressible): (vsim-FLI-3155) The FLI is not enabled in this version of ModelSim.
# End time: 16:15:14 on Dec 31,2019, Elapsed time: 0:00:00
# Errors: 3, Warnings: 1
make[1]: *** [results.xml] Fout 211
make[1]: Map '/tmp/cocotb_modunderscore' wordt verlaten
make: *** [sim] Fout 2
```
When I remove ``dut.dff._discover_all()`` from _test.py_; the simulation passes without segfault.

garmin-mjames, 2020-01-28:
This looks to be the culprit:
> ``` # ** Error (suppressible): (vsim-FLI-3155) The FLI is not enabled in this version of ModelSim.```

ModelSim uses `FLI` to access VHDL.

themperek, 2020-01-28:
@Fatsie Please close if the issue is solved.

closed 2020-12-03 by github-actions[bot]

stdefeber, 2023-04-07:
This issue is not as dead as it appears.
I have a similar issue. However this involves xcelium.

OS : CentOs 7.6
CoCoTb : 1.6.2
Xcelium : 21.03.008

We switched from having mostly VHDL as top and intermediate layers to verilog/SV.
When a VHDL module is compiled into a library, CoCoTb can not find it anymore.

`CURDIR=$(realpath .)

TOPLEVEL := tb_top

COCOTBMAKEDIR=$(shell cocotb-config --makefiles)

VHDL_SOURCES_module_lib := \
	$(CURDIR)/../RTL/dff.m.vhdl \
	$(CURDIR)/../INTERFACE/sub.e.vhdl \
	$(CURDIR)/../RTL/sub.a.vhdl

VHDL_SOURCES := \
	$(CURDIR)/../RTL/sub_wrapper.vhdl \

VERILOG_SOURCES :=  \
	$(CURDIR)/../RTL/core.sv \
	$(CURDIR)/../RTL/ic.sv \
	$(CURDIR)/../RTL/top.sv \
	$(CURDIR)/../TESTBENCH/tb_top.v

VHDL_LIB_ORDER = \
    module_lib \

export COCOTB_LOG_LEVEL=DEBUG
TOPLEVEL_LANG=verilog
MODULE=test
SIM=xcelium
COCOTB_HDL_TIMEUNIT=100fs
COCOTB_HDL_TIMEPRECISION=100fs
COCOTB_RESOLVE_X=ZEROS
COMPILE_ARGS += -v93 -V200X -accessreg +rwc $(INCDIRS)

include $(COCOTBMAKEDIR)/Makefile.inc
include $(COCOTBMAKEDIR)/Makefile.sim`

stdefeber, 2023-04-07:
A work around has been found.
When all sources are marked as VHDL_SOURCES it works. 

```
VHDL_SOURCES_module_lib := \
	$(CURDIR)/../RTL/dff.m.vhdl \
	$(CURDIR)/../INTERFACE/sub.e.vhdl \
	$(CURDIR)/../RTL/sub.a.vhdl

VHDL_SOURCES := \
	$(CURDIR)/../RTL/sub_wrapper.vhdl \
	$(CURDIR)/../RTL/core.sv \
	$(CURDIR)/../RTL/ic.sv \
	$(CURDIR)/../RTL/top.sv \
	$(CURDIR)/../TESTBENCH/tb_top.v

```

Nabih-Saleh, 2024-02-27:
unfortunately, this work around does not always work!, because it complain when compiling .v file because it found illegal syntax based on vhdl compiler