cocotb/cocotb #1275: Access SV structs as HierarchyObject on verilator

nndurj, 2020-01-01:
On verilator, SV structs are shown as  `ModifiableObject`. Because of this it shows up a single dimensional packed vector, and we can't access the struct fields separately. On questasim, they are shown as `HierarchyObject`, and we can access the fields.

### Test Program

```systemverilog
module t();

struct packed{
    logic a;
    logic b;
} hdr;

endmodule
```
[image]

wallento, 2020-01-03:
This is certainly a Verilator issue, I am just adding vpi support on the fly and this is probably not handled as queried by cocotb. I will trace it and create a standalone test for Verilator.

wallento, 2020-01-12:
This seems harder than I hoped as structs are not handled at all right now (https://github.com/verilator/verilator/issues/860). After that I can access it with VPI which will be another larger task. This will probably take a while..

nanduraj1, 2020-04-21:
There is a switch `--trace-structs` that shows struct/union hierarchy in vcd file. The code snippet is below. Is it possible to do something similar for vpi?  

https://github.com/verilator/verilator/blob/174fd1bf0ed5256cf62aa1cebf0774aafb4ff1cc/src/V3TraceDecl.cpp#L306-L341

imphil, 2020-05-23:
@nanduraj1 This is a missing functionality in Verilator. You need to work with the Verilator project to get this functionality added; once it is there and implemented according to the VPI standard, cocotb shouldn't be able to use it immediately.

imphil, 2020-05-23:
Closing this for now as there's nothing we can do on the cocotb side.

ktbarrett, 2025-10-16:
Support for this was removed, so this is a noop.

ktbarrett, 2026-10-23:
Actually this is still planned and actually Verilator supports this as of 5.050.

closed 2026-10-23 by ktbarrett