cocotb/cocotb #1277: Release cocotb 1.3

imphil, 2020-01-02:
- [ ] prepare release notes #1270 
- [ ] test packaged release from Ci 
- [ ] prepare release announcement email 
- [ ] tag release 
- [ ] test release build on pypi
- [ ] send out announcement to mailing list

imphil, 2020-01-03:
RC1 is tagged and building at https://travis-ci.org/cocotb/cocotb/builds/632364151

eric-wieser, 2020-01-03:
There are a few PRs left in the milestone that we should either merge or unmilestone. I'll have a look at one of them, #1262 shortly.

imphil, 2020-01-03:
Let's limit ourselves to (important) regressions over 1.2.0; I didn't see any of those, but you have been following more PRs recently, so please let me know if I missed anything. We can always do more release candidates if necessary.

imphil, 2020-01-03:
Looks like https://github.com/cocotb/cocotb/pull/751 didn't get merged in time, even though it is ready. @Matthewar I'm sorry -- how do you feel about getting this into 1.2? We can do a rc2 if you'd like to get it in.

eric-wieser, 2020-01-03:
@imphil, just merged that. I assume you mean 1.3?

imphil, 2020-01-03:
I meant 1.3, yes. Updated above. Since we have now changes on top of rc1 we need to do a rc2 anyways. @eric-wieser I'll do a rc2 once you're happy with the current state (please let me know here). After that, please don't merge anything after that point until we have a final release to avoid invalidating the testing done on the RC build.

imphil, 2020-01-03:
rc1 is now on pip: https://pypi.org/project/cocotb/1.3.0rc1/

As said before, I'll wait for rc2 before doing a "request for testing" on the mailing list (most likely early tomorrow).

eric-wieser, 2020-01-03:
I plan to merge no more PRs unless I get your approval on them.

The only remaining [tagged bug PRs](https://github.com/cocotb/cocotb/pulls?q=is%3Aopen+is%3Apr+label%3Atype%3Abug) as of writing are #1256 (draft), #1262, and #1268.

If you don't think 1.3 needs any of these, then please go ahead with the release!

imphil, 2020-01-04:
1.3.0rc2 is ready, request for testing sent to the mailing list at https://lists.librecores.org/pipermail/cocotb/2020-January/000072.html. I intend to release 1.3.0 final mid next week if nothing unexpected happens. Please label everything blocking the release with the 1.3.0 milestone, and don't check in any other changes until then to avoid invalidate the testing done on rc2.

imphil, 2020-01-08:
Release is out!

Release notes: https://cocotb.readthedocs.io/en/latest/release_notes.html#cocotb-1-3-0
Release announcement: https://lists.librecores.org/pipermail/cocotb/2020-January/000073.html

closed 2020-01-08 by imphil

cmarqu, 2020-01-08:
Would now be a good time to set the version to ``2.0-pre``?

imphil, 2020-01-08:
I'd go for 1.4 for now, not 2.0, until we agree on including breaking changes (either regarding the Python version, or other features)

Looking at https://www.python.org/dev/peps/pep-0440/#pre-releases, we'd probably end up with 1.4a1?

Any opinion @eric-wieser @themperek ?

eric-wieser, 2020-01-09:
> Any opinion @eric-wieser @themperek ?

My opinion is we should drop support for 2.7 in master immediately, and perform all the simplifications that entails.

Assuming we do that, our choice is either:
1. Commit to never releasing new _features_ for 2.7 support. Critical bugfixes can still be published as 1.3.1 (kept on a `maintenance/1.3.x` branch), but I suspect in practice we won't end up doing that. Either way, we can now use 1.4 for our first Py3-only release. I would argue dropping support for a python version isn't really a breaking change, the natural release cycle of python means that every year or so it's likely we drop the oldest version.
2. Leave the door open for new 2.7 feature releases. This means reserving `1.x.x` numbers for those releases, so master would need to become `2.0,x`.

My preference here is for 1 - we struggle to keep up with PRs as is, we certainly can't afford to maintain two versions right now.

themperek, 2020-01-09:
Dear All,

I am in favor of Eric's proposal. As I understand if we drop python 2
support we can clean up the code significantly and make it faster.
About numbering, we can decide something and change later before release.
We should probably make 1.3.x.

If you think it is useful I can organize a small telephone meeting to maybe
discuss some plans in person?

Regards,
Tomek

imphil, 2020-01-09:
> If you think it is useful I can organize a small telephone meeting to maybe
discuss some plans in person?

Sounds good. If anyone reading this issue wants to join send me a quick note (I'm already counting Eric and Tomasz in). I'll try to find a timeslot tomorrow.