cocotb/cocotb #1278: adder_module test doesn't run properly

eric-wieser, 2020-01-02:
From a random CI run:
```
/c/iverilog/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s adder -g2012   C:/src/examples/adder/tests/../hdl/adder.v
PYTHONPATH=/c/src/build/libs/x86_64:/c/src/examples/adder/tests:/:/C/c/srcC/src/examples/adder/model:/C/src/examples/adder/tests:/C/miniconda3/python37.zip:/C/miniconda3/DLLs:/C/miniconda3/lib:/C/miniconda3:/C/miniconda3/lib/site-packages:/C/miniconda3/lib/site-packages/cocotb-1.2.0-py3.7.egg:/C/miniconda3/lib/site-packages/win32:/C/miniconda3/lib/site-packages/win32/lib:/C/miniconda3/lib/site-packages/Pythonwin PATH=/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Windows/System32/OpenSSH:/c/Users/ContainerAdministrator/AppData/Local/Microsoft/WindowsApps:/c/miniconda3:/mingw-w64/bin:/usr/bin:/bin:/c/miniconda3/Scripts:/c/miniconda3/bin:/c/miniconda3/condabin:/c/iverilog/bin:/c/iverilog/gtkwave/bin:/c/iverilog/bin:/c/src/build/libs/x86_64 MODULE=test_adder \
        TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /c/iverilog/bin/vvp -M /c/src/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Icarus Verilog version 10.1 (stable)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.2.0 from C:/src
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1577976144
     0.00ns CRITICAL cocotb.regression                         regression.py:137  in initialise                      Failed to import module test_adder: No module named 'adder_model'
     0.00ns INFO     cocotb.regression                         regression.py:138  in initialise                      MODULE variable was "test_adder"
     0.00ns INFO     cocotb.regression                         regression.py:139  in initialise                      Traceback: 
     0.00ns INFO     cocotb.regression                         regression.py:140  in initialise                      Traceback (most recent call last):
                                                                                                                       File "C:\miniconda3\lib\site-packages\cocotb-1.2.0-py3.7.egg\cocotb\regression.py", line 135, in initialise
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "C:\miniconda3\lib\site-packages\cocotb-1.2.0-py3.7.egg\cocotb\regression.py", line 63, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "C:\src\examples\adder\tests\test_adder.py", line 5, in <module>
                                                                                                                         from adder_model import adder_model
                                                                                                                     ModuleNotFoundError: No module named 'adder_model'
                                                                                                                     
     0.00ns ERROR    cocotb.gpi                                  gpi_embed.c:425  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:198  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
make[3]: Leaving directory '/c/src/examples/adder/tests'
```

cmarqu, 2020-01-02:
Works for me locally (with Xcelium).

eric-wieser, 2020-01-02:
Looks like a makefile issue, `PYTHONPATH` is garbled:

 
 
 
 PYTHONPATH=/c/src/build/libs/x86_64:/c/src/examples/adder/tests:/:/C/c/srcC/src/examples/adder/model

cmarqu, 2020-01-02:
The same with Linux works: https://travis-ci.org/cocotb/cocotb/jobs/631880541

cmarqu, 2020-01-02:
Maybe for a new issue: this doesn't get marked as failed in Travis - do we want/expect that behaviour?

elgorwi, 2020-01-03:
I think this is related to different path separators on linux and windows platforms. Maybe something has changed in the way the $PYTHONPATH variable is set. I’m mostly using linux these days, but I’m pretty sure this example used to run correctly on windows platform as well.

themperek, 2020-01-21:
@eric-wieser This may help: https://github.com/cocotb/cocotb/pull/1099/files#diff-43e017b28e778fbb645eb72746a5c1adR70

eric-wieser, 2020-01-22:
> Maybe for a new issue: this doesn't get marked as failed in Travis

I think @ktbarrett PR will fix that.