cocotb/cocotb #1289: Python 2 Cleanup tracking issue

eric-wieser, 2020-01-09:
A list of cleanups we can make when dropping python 2:

* [ ] Delete https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/simulator/simulatormodule_python2.c, and merge https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/simulator/simulatormodule_python3.c with https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/simulator/simulatormodule.c

* [ ] Replace `reject_remaining_kwargs` with direct syntax
* [ ] Combine `cocotb_test` and `cocotb_test_35` testbenches
* [ ] Removes internal uses of `cocotb./_py_compat`
* [ ] Remove `cocotb.outcome` hacks for python 2 (re `__traceback__`)s
* [ ] Change all library coroutines to use `async def`, for speed

Relates to #767. Maintainers, please add items to this list / edit the list to link PRs.

cmarqu, 2020-01-09:
Not all of those are strictly cleanup:
- Remove ``(object)`` from ``def MyClass(object):``
- Use ``__class__.__qualname__`` instead of ``__class__.__name__``
- Add type annotations (for user-facing code)

eric-wieser, 2020-01-09:
@cmarqu: Added those to the list, thanks!

cmarqu, 2020-01-09:
You are too quick, edited in one more line :)

cmarqu, 2020-01-09:
- Remove 2.7 version from Travis CI setup

cmarqu, 2020-01-09:
- Change minimum supported version in documentation
- Adjust https://github.com/cocotb/cocotb/blob/9caf77cb6ceae739858ecd4a3760631470aac77c/setup.py#L59

cmarqu, 2020-01-12:
- Type annotations from docstrings can be removed provided we (a) have them on Python level and (b) we use (the yet unreleased) version of Sphinx 2.4 containing https://github.com/sphinx-doc/sphinx/pull/7018

garmin-mjames, 2020-01-13:
Remove `PY_MAJOR_VERSION < 3` code in `gpi_embed.c`.

ktbarrett, 2020-01-27:
Remove reference from `tox.ini`.

ktbarrett, 2020-03-11:
Remove all `__nonzero__` functions.

ktbarrett, 2020-03-11:
Fun times ahead. I was rewriting test timeout to use async functions, which required rewriting `with_timeout` to handle async functions, which is most reasonably accomplished as such:

```
async def with_timeout(trigger, timeout_time, timeout_unit=None):

    async def _timeout():
        await cocotb.triggers.Timer(timeout_time, timeout_unit)
        raise cocotb.result.SimTimeoutError()

    return await First(_timeout(), trigger)
```

`First` in this case would `fork` `_timeout()` (or I could here, regardless...). But the timeout exception will only propogate if #922 is resolved first.

eric-wieser, 2020-03-19:
@ktbarrett, is that something that would be fixed by changing `First` to also be `async def`?

ktbarrett, 2020-03-19:
@eric-wieser Yes, in fact I already have that change on a branch. I missed it originally.

eric-wieser, 2020-03-20:
Updated the top-matter to group by difficulty, so new contributors don't pick up a hard task by accident.

ktbarrett, 2020-03-25:
How should we deal with adding type annotations while still supporting 3.5? Many interesting (to us) types aren't available on all version of 3.5, e.g. `Coroutine`.

eric-wieser, 2020-03-27:
@ktbarrett: An easy option would be to add a dependency on the `typing` module from pypy, which backports the newer annotations.

Are you sure we need `Coroutine`? My impression was that on `async` functions it was implied.

eric-wieser, 2020-03-27:
I'd lean towards adding all the easy type annotations first anyway though.

ktbarrett, 2020-03-27:
@eric-wieser I was documenting the interfaces on #1526 when I ran into this, which is why it has no annotations.

```python
def fork(coro: Union[Coroutine, cocotb.decorators.RunningTask]) -> cocotb.decorators.RunningTask:
    # ...
```

eric-wieser, 2020-03-27:
@ktbarrett: `collections.abc.Coroutine` was added in 3.5: https://docs.python.org/3/library/collections.abc.html#collections.abc.Coroutine

Sure, maybe you can't use `typing.Coroutine`, but it doesn't really matter.

ktbarrett, 2020-03-27:
You're right, we aren't ever going to constrain those instances. But I have a feeling this will be an ongoing problem. We saw the same issue other week with `importlib-resources` (they have since added patch versions to the `typing` documentation :^).

ktbarrett, 2020-04-05:
Move to format string language instead of using old modulus string formatting.

ktbarrett, 2020-04-05:
I don't know if this is really a Python 2 cleanup issue, but use `type(self)` instead of `self.__class__`.

eric-wieser, 2020-04-05:
Strictly neither that nor the `.format` calls are python 2 cleanup.

ktbarrett, 2020-04-05:
Did they renege on the removal of the modulus syntax in Python 3? I thought it was deprecated and "to be removed"?

eric-wieser, 2020-04-05:
@ktbarrett: Pretty sure they did, especially since `b'{}'.format(1)` is an `AttributeError`, but `b"%d" % 1` isn't.

eric-wieser, 2020-04-05:
I take it back, `type(x)` _is_ a python 2 cleanup, since it didn't work correctly on old-style classes.

eric-wieser, 2020-04-29:
Thanks all for assembling this list. I think it's now outlived it's usefulness now that it's mostly complete, so I've extracted remaining tasks to standalone issues.

closed 2020-04-29 by eric-wieser