cocotb/cocotb #1294: Dropping into the cocotb REPL

BracketMaster, 2020-01-13:
[Here](https://cocotb.readthedocs.io/en/latest/quickstart.html#reading-values-from-signals) there is an example like this where you can interact with your design from the REPL:

```python3
>>> # Read a value back from the DUT
>>> count = dut.counter.value
>>>
>>> print(count.binstr)
1X1010
>>> # Resolve the value to an integer (X or Z treated as 0)
>>> print(count.integer)
42
>>> # Show number of bits in a value
>>> print(count.n_bits)
6
```

I'm able to do this with: ``import code; code.interact(local=locals())`` as shown in the following example:
```python3
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def my_first_test(dut):
    """Try accessing the design."""

    dut._log.info("Running test!")
    for cycle in range(10):
        dut.CLK_IN = 0
        yield Timer(1, units='ns')
        dut.CLK_IN = 1
        yield Timer(1, units='ns')
    dut._log.info("Running test!")
    import code; code.interact(local=locals())
```

Is this the write way to do that - is their perhaps some flag I can pass in the cocotb makefile?

ktbarrett, 2020-01-13:
That works but is very limited. You can't progress simulation time that way, only touch the dut then exit. We are currently working towards an interactive testing, but it is not currently supported.

BracketMaster, 2020-01-13:
OK. No worries. 

And one more question, verilator has the concept of tb.eval() - that is, I make a bunch of model updates and then evaluate.

Are Triggers the same concept in cocotb? That is, after doing multiple DUT assignments such as ``dut.sig = 1`` etc. does calling ``yield Timer()`` then "evaluate the model"?
This would also mean that all the previous assignment statements such as ``dut.sig = 1`` did not force a model evaluation...

Is that correct?

And lastly, how can I check the current time in the simulation? As in, can I do something like ``print(f"TIME {dut.get_time(unit-'ns')}ns")``?

ktbarrett, 2020-01-13:
>Are Triggers the same concept in cocotb? 

Not exactly. Triggers are `yield`ed or `await`ed and used to block the current coroutine execution, allowing the simulator to progress time, until the Trigger's simulation event occurs. If you wait for `RisingEdge(dut.clk)` the simulator will run until `dut.clk` goes from low to high, at which point it will jump back into Python at the point you `yielded` on the `RisingEdge` trigger and continue executing that coroutine.

>That is, after doing multiple DUT assignments such as dut.sig = 1 etc. does calling yield Timer() then "evaluate the model"?

Simulators model hardware. In real life, computations take time, so I think this is just thinking about it wrong. If you progress time, only what can happen in that amount of real time will happen. If you wait 10 ns and your clock rate is 50MHz (20 ns period), nothing registered will be computed yet, however combinational circuits will be.

>This would also mean that all the previous assignment statements such as dut.sig = 1 did not force a model evaluation... Is that correct?

This is correct. When you assign a value to a signal, it does not take it's value immediately, but only after time has passed in the simulator.

>And lastly, how can I check the current time in the simulation?

[`cocotb.utils.get_sim_time`](https://cocotb.readthedocs.io/en/latest/library_reference.html#cocotb.utils.get_sim_time).

If you have any other question, you can post them in our [gitter](https://gitter.im/cocotb/Lobby).

closed 2020-01-13 by BracketMaster